raulmur/ORB_SLAM #2: test on KITTI VO seq 00: Not initialized -> Trying to initialize

amiltonwong, 2015-03-17:
Dear all,

I had installed ORB_SLAM and run the example successfully. Now I turn into having a test on KITTI VO seq 00, like authors did in their paper. 
But when I start ORB_SLAM on KITTI VO seq 00, I just receive "Not initialized -> Trying to initialize", and seeing street-view images with green edges on it in the /ORB_SLAM/Frame, please the link:
https://www.dropbox.com/s/0oi3iu3nqzwpv1h/47.png?dl=0

Could someone give me some hints?

Thanks in advance~

Milton

raulmur, 2015-03-17:
Are you loading the correct camera calibration? (settings file)

amiltonwong, 2015-03-18:
@raulmur ,
According to the calib file of KITTI VO seq 00 
https://www.dropbox.com/s/pvmavyymbmwir3h/calib.txt?dl=0
I modify the Settings.yaml as follows:
https://www.dropbox.com/s/l8r81vxz29ehiys/Settings-kitti.yaml?dl=0

============================================================

%YAML:1.0

\# Camera Parameters. Adjust them!

\# Camera calibration parameters (OpenCV) 
Camera.fx: 718.856
Camera.fy: 718.856
Camera.cx: 607.1928
Camera.cy: 185.2157

\# Camera distortion paremeters (OpenCV) --
Camera.k1: 0
Camera.k2: 0
Camera.p1: 0
Camera.p2: 0

\# Camera frames per second 
Camera.fps: 10.0

\# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

=======================================================================

Any comments on the Settings-kitti.yaml file or suggestions?

Thanks~

raulmur, 2015-03-18:
The calibration parameters are ok.  
The frame you attached in your first post (https://www.dropbox.com/s/0oi3iu3nqzwpv1h/47.png?dl=0) seems no to have the correct size. It should be 1241x376.  Have you modified the size of the original images?

amiltonwong, 2015-03-19:
@raulmur ,
I had turned back to original resolution : 1241x376, but the result is not satisfied. Please see the following link:
http://goo.gl/XVWwMQ
After the ORB_SLAM tracks the first 15 seconds, it loses the tracking at turning the corner. And then relocalizes itself back only in last 15 seconds. In the large part duration, it only shows  "TRACK LOST".

Any comments or suggestions?

Thanks~

raulmur, 2015-03-19:
There could be different reasons...
- The fps/timestamps of your rosbag are correct? It should be 10 fps.
- Your computer is not powerful enough for this sequence. This dataset is really exigent.
- In the paper we say that we extract 2000 ORB in KITTI sequences, because of the resolution of the images. Still with 1000 should work.

amiltonwong, 2015-03-19:
@raulmur ,
1. my command for generating bag file is as follows:
 rosrun BagFromImages BagFromImages /root/Desktop/KITTI/00/image_2/ .png 10 /root/Desktop/KITTI/00/image_2.bag
so , it should be 10fps.
2. my computer configuration is CPU: i3 , memory: 4GB ram, what preferred configuration will you suggest?
3. I modify the Settings.yaml file, now use 2000 for nfeatures, but the result doesn't improve, still lose tracking at first turning the corner.

raulmur, 2015-03-19:
I suggest to use the grayscale images (image_0)  and we use a i7-4700MQ computer.
The system runs 3 threads in parallel (trackins, local mapping and loop closing)., I would say the i3 is not enough for this dataset.

amiltonwong, 2015-03-21:
Yes, @raulmur, you're right. I change to use an i7-3520M cpu laptop, and the result is satisfied as showed in your demo.
For testing on 00 seq, around at 290s (its whole length is 454 seconds), the [rviz-3] process has died, but the tracking still continues till the end. For log snapshot, you can see in the following link. 

https://www.dropbox.com/s/cqrlwijiq2nfop4/log_snapshot.png?dl=0

Any comments or suggestions to fix it?

THX~

raulmur, 2015-03-23:
That is a problem of the visualizar rviz, and I cannot help you with that. If that happens just open rviz again.

closed 2015-03-23 by raulmur