raulmur/ORB_SLAM #8: Do we need to change the Cmake file  to  catkin before compiling  in hydro

shabhu18, 2015-03-30:
The Cmake file provided with the code is useful for only compiling in ros fuerte .Can you provide the files for hydro version.

raulmur, 2015-03-30:
The same cmake file works for hydro. Have you set up the ROS_PACKAGE_PATH environment variable?

shabhu18, 2015-04-01:
I have copied my ORB SLAM folder to catkin_ws .After buiding all the files in third party. I am getting this error
https://drive.google.com/file/d/0Bz3BfhMrLdoyd1dOUzluZ2swTzg/view?usp=sharing

NikolausDemmel, 2015-04-01:
It is a rosbuild package, not catkin: http://wiki.ros.org/catkin/Tutorials/using_rosbuild_with_catkin

bmagyar, 2015-04-08:
Hey guys, I have catkinized the packages and will submit a pull request with it soon.
https://github.com/bmagyar/ORB_SLAM/tree/indigo-devel
Btw it is tested with both hydro and indigo.

mryellow, 2015-04-15:
+1 came here to post issue and get started started on PR to make this usable.... Nice to see it's already done.

bmagyar, 2015-04-15:
#12

KathiravanNatarajan, 2017-02-03:
@shabhu18 : instead of keeping it in src folder, keep it in catin_ws itself and give the environment variable like this : 
`export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:~/catkin_ws`
and don't forget to run this.
`source  ~/.bashrc`