raulmur/ORB_SLAM #9: Always" not initialize"

liqile, 2015-04-01:
Dear Raul Mur
Hello, this is Li Qile
    I am running your code with my own data recently. but the orb slam cannot initialize, I can only see green lines on the `/ORB_SLAM/Frame`
    May I ask you for some suggestions to fix such situsations?
    Thanks a lot
     Best
                                                                              Li Qile

raulmur, 2015-04-01:
Firstly check that you are loading the correct calibration data.
In order to initialize, the camera must translate enough while looking at the same scene. 
Rotations should be avoided during initialization.

liqile, 2015-04-03:
@raulmur 
hello, my camera is 1280 x 720
 how do I change your code to fix it ?
thanks

raulmur, 2015-04-03:
You do not need to change the code.
You just need to modify the Settings file with the calibration of your camera.
To calibrate the camera you can use OpenCV (http://docs.opencv.org/doc/tutorials/calib3d/camera_calibration/camera_calibration.html) or the Matlab Calibration Toolbox (http://www.vision.caltech.edu/bouguetj/calib_doc/)

closed 2015-04-03 by raulmur