raulmur/ORB_SLAM #13: How to create ORB vocabulary from dataset?

bjornph, 2015-04-16:
While I have managed to get your example file to work, I struggle with providing an ORB vocabulary from my own dataset. I have a set of image files converted to a .bag file. 

How do you extract ORB features from an image dataset to provide to ORB-SLAM?

raulmur, 2015-04-17:
You do not need to create another visual vocabulary. The vocabulary we provide was taken with a very big dataset with indoor and outdoor images. We have always used the same vocabulary in our experiments achieving very good performance. If you are still interested in creating your own, you have to use the DBoW2 library (https://github.com/dorian3d/DBoW2).

closed 2015-04-17 by raulmur

amc-nu, 2015-06-17:
Raul, could you tell us which image dataset did you use to create the dictionary?

Gracias!

jingpang, 2015-12-12:
@amc-nu , the dataset is here (http://www.rawseeds.org/rs/datasets/view//7) , 'Bovisa_2008-09-01',
according to this paper by the author 'Fast Relocalisation and Loop Closing in Keyframe-Based SLAM
'. But i failed to download it...

KathiravanNatarajan, 2017-02-03:
what does ORB vocabulary mean ? I am new to this . if someone explains me, it'd be great

KathiravanNatarajan, 2017-02-03:
can we implement this ORB_SLAM for ARDRONE ? should I have to make any changes for that ?

chaiein, 2017-02-09:
you can get answer here https://www.zhihu.com/question/49153462.

KathiravanNatarajan, 2017-02-10:
@chaiein  : Hi,
I have got rviz(something is running) and rqt_image window with the output of ardrone/front/image_raw.

How can I get the 3D model of the environment using the existing code ? I have downloaded bag file too. Does bag file have anything to do with 3D model ? Does the code already contain a way to generate the 3D model ? 
It'll be a great help if you can answer this. I couldn't get help from anyone.

manthan99, 2020-07-16:
Please checkout this repo where I have added step by step instructions to create own vocabulary for ORB SLAM
https://github.com/manthan99/ORB_SLAM_vocab-build

KlemenDEV, 2021-01-21:
Do we need to prepare our dataset from un-distorted camera images, or does this not matter?

TonyPhh, 2021-12-09:
> @amc-nu , the dataset is here (http://www.rawseeds.org/rs/datasets/view//7) , 'Bovisa_2008-09-01', according to this paper by the author 'Fast Relocalisation and Loop Closing in Keyframe-Based SLAM '. But i failed to download it...

hello，can you download the ‘Bovisa_2008-09-01’ dataset? I get some trouble and have difficulty downloading the 'Bovisa_2008-09-01' dataset. thanks!