raulmur/ORB_SLAM #14: Fail to initialize the relative pose of first two frames

liqile, 2015-04-19:
Hi, @raulmur r
I am sorry to trouble you but I am facing a very hard problem
I was using your method to compute the relative pose of the first two frames.
but I always get two solutions, and the value of "bestGood" and "secondBestGood" nealy the same.
I am looking forward to hearing from you and best wishes to you.
thanks in advance

                                                        sincerely
                                                           Li Qile

raulmur, 2015-04-19:
If you have never been able to initialized with ORB-SLAM using your camera and trying it out with different movements and environments, then your are not providing the system with the correct camera calibration. During initialization avoid rotations and perform translations so that the camera is always looking at the same scene.

closed 2015-04-19 by raulmur

liqile, 2015-04-19:
@raulmur , but the bestGood and the secondBestGood can reach the amount of inlier points..