raulmur/ORB_SLAM #15: Could you provide different kinds of ORB vocabulary

jim1993, 2015-04-20:
I have installed ORB_SLAM on the odroid u3.
However, the ORBvoc file is too big to run on the odroid board.
I would appreciate it if you could offer another smaller ORB vocabulary file just for indoor environment.

gitunit, 2015-04-20:
as **raulmur** already mentioned in another topic, you can create your own vocabulary using this tool https://github.com/dorian3d/DBoW2

jim1993, 2015-04-21:
Thank you.
I have successfully created a new smaller vocabulary file and run it on the odroid u3.

closed 2015-04-21 by jim1993

raulmur, 2015-04-21:
Good to here you made it working. 
If your vocabulary tree contains less than 4 levels you will need to change the 4 in the instruction mpORBvocabulary->transform(vCurrentDesc,mBowVec,mFeatVec,4);
in function KeyFrame::ComputeBoW  (Keyframe.cc)
and in Frame::ComputeBoW (Frame.cc)
That instruction associates features to nodes in the 4th level of the vocabulary (which by default contains 6 levels) and store that info in mFeatVec to faciliate feature matching in several tasks. 
I should make this clear in the documentation...

jingpang, 2015-12-12:
@jim1993  I failed to download the dataset used by raulmur. Which dataset did you use to create the vocabulary and how about the performance when running on the odroid U3? Would you please share the vocabulary file?

Thank you.

jim1993, 2015-12-15:
Hi, @jingpang 
As gitunit mentioned above, you can create your own vocabulary file from: 
https://github.com/dorian3d/DBoW2

About the performance, it depends on the size of your vocabulary file.
I suggest use Odroid XU3 or XU4 instead of U3 if you want to get acceptable result.
(Since the RAM of U3 is not big enough)

ps. My previous notebook was broken, I don't have the back file, sorry.

jingpang, 2015-12-16:
@jim1993 Thank you for your reply!
I know the method to create. However I can't download the image dataset "Bovisa_2008-09-01" which are used by the author. (http://www.rawseeds.org/rs/capture_sessions/view/10)

So I wonder which dataset did you use, do you have the link? Thanks again!

jonas-, 2015-12-16:
Hi guys, I'm trying to create my own ORB vocabulary and wonder if you could share the used code to create the vocabulary.

Thanks, Jonas

alejodosr, 2016-02-02:
Hi @jim1993 , How did you manage to obtain a *.txt (¿serialized?) vocabulary file? I have used the DBow2 demo to generate the vocabulary (*.yml) but the ORB-SLAM2 algorithm does not recognize the file.

jim1993, 2016-02-02:
Hi @alejodosr 
Sorry, since it was just one test method of my final project and I already graduated, I can not find any backup file or memo. However, I am pretty sure that you can easily create your own ORB vocabulary by following dorian3d's tutorial of "DBoW2" -->https://github.com/dorian3d/DBoW2
(The main source codes of "DBoW2" are still the same as last time I successfully used it.)

jonas-, 2016-02-11:
@alejodosr just returned form my holidays and uploaded the code to github:
https://github.com/jonas-/myStuff/blob/master/createORBVoci.cpp

Hopefully it may be useful for you.
It's the code I used without modifications to create a vocabulary file from a ROS bag file.
The code needs dependencies to DBoW2 and the ORBextractor from ORB-SLAM.

alejodosr, 2016-02-11:
@jonas- thank you very much! You've saved me a lot of time of research. I'll try it out and test if the vocabulary was my actual issue.

sharma-arjun, 2017-04-30:
@jonas- Thanks for the file. How do I compile it?

jonas-, 2017-05-01:
@sharma-arjun make sure you have all the dependencies in the header installed and correctly linked (i.e. DBoW2, ORBextractor from ORB-SLAM, opencv, ros).
I use to compile everything with CMakeLists.txt files