raulmur/ORB_SLAM #17: error:undefined symbol

liqile, 2015-04-22:
hi, I am running your code with dataset from tum-vision
but an error occured, and the process died.

`[ INFO] [1429678011.002510623]: New Map created with 425 points
/home/hongbohuang/catkin_ws/src/ORB_SLAM/bin/ORB_SLAM: symbol lookup error: /home/hongbohuang/catkin_ws/src/ORB_SLAM/bin/ORB_SLAM: undefined symbol: _ZN3g2o17EdgeSE3ProjectXYZC1Ev`
`[ORB_SLAM-3] process has died [pid 13939, exit code 127, cmd /home/hongbohuang/catkin_ws/src/ORB_SLAM/bin/ORB_SLAM Data/ORBvoc.yml Data/Settings.yaml __name:=ORB_SLAM __log:=/home/hongbohuang/.ros/log/6cfb0986-e893-11e4-be96-bcaec584f5df/ORB_SLAM-3.log].
log file: /home/hongbohuang/.ros/log/6cfb0986-e893-11e4-be96-bcaec584f5df/ORB_SLAM-3*.log`

could you please tell me what's wrong with my options? thanks in advance
best

raulmur, 2015-04-22:
I defined EdgeSE3ProjectXYZ in the g2o version I provide with ORB-SLAM. It seems you are not correctly linking to that g2o version.

liqile, 2015-04-26:
hi, @raulmur when I run lsd-slam, I installed the dependencies like "`ros-indigo-libg2o`", will it influence the g2o used in orbslam?

liqile, 2015-04-26:
@raulmur when I first ran orb-slam, no such error happened, but after I installing and running lsd-slam, I could not running your orb-slam any longer

raulmur, 2015-04-26:
The problem is that you are not linking to my g2o version, but "ros-indigo-libg2o".

liqile, 2015-04-26:
@raulmur e.. how to linking to your g2o version?
recompile orb-slam again?

husdo, 2015-05-06:
@liqile Did you managed to make it work? I have exactly the same problem. (LSD-SLAM) Thanks

liqile, 2015-05-07:
@husdo 
you can try the command:`sudo apt-get remove ros-indigo-libg2o`
the ros distribution of mine is `indigo`, you should replace it with your ros distribution
best