raulmur/ORB_SLAM #18: error

shabhu18, 2015-04-23:
I was running ORB SLAM on my own  data set , I got this error.

terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
[ORB_SLAM-3] process has died [pid 7918, exit code -6, cmd /home/shabhu/Work/rosfuerte/ORB_SLAM/bin/ORB_SLAM Data/ORBvoc.yml Data/Settings.yaml __name:=ORB_SLAM __log:=/home/shabhu/.ros/log/28136f02-e8d9-11e4-bce3-18a905b9cd56/ORB_SLAM-3.log].
log file: /home/shabhu/.ros/log/28136f02-e8d9-11e4-bce3-18a905b9cd56/ORB_SLAM-3*.log

Can you help me out?

raulmur, 2015-04-23:
Run a debugger and check where it is crashing. Without more info I cannot tell what is happening.

vibhav-vineet, 2015-05-08:
hi Shabhu, hi raulmur, 

I am also facing the issue. 

ORBSLAM works fine for the example sequence provided by raulmur and on TUM dataset but I start getting this error on the KITTI dataset. 

Let me know if you figured out the issue. 

Thanks,