raulmur/ORB_SLAM #21: rviz crash

gitunit, 2015-04-26:
hola raúl,
did you ever run into the following error? http://answers.ros.org/question/192272/possible-limitation-of-points-in-markerline_strip-and-rviz/

by now i had this problem several times, mostly this happened after loop closure when my rosbag run the second time. i know its a rviz problem but maybe you solved this already?

raulmur, 2015-04-26:
Sorry, but I never had that error.

gitunit, 2015-04-26:
ok. thanks. further searching revealed its a known issue in rviz. https://github.com/ras-sight/hratc2015_framework/issues/4

also i realised that despite this error, ORB_SLAM is still running in the background and one can simply open a new console window and re-run rviz and open the provided config file. then everything is back again. :)

gitunit, 2015-04-26:
i've tested a little bit more and realized that this crash always happens when KFs reach 1024. then i reopen rviz again (as mentioned earlier, ORB_SLAM is still running) then the next chrash happens exactly when reaching KFs of 2048.
i thought this information maybe useful.

lilouch, 2015-04-28:
@gitunit  I have also the same issue ! Did you find a way to solve this ?

gitunit, 2015-04-28:
no, i haven't. i tried a self compiled version of rviz which is build on ogre 1.10 but still the same issue.

HChengmeng, 2015-06-12:
@gitunit  @lilouch  I have  the same issue ! Did you find a way to solve this ?

lilouch, 2015-06-13:
No still didn't find a solution..

gitunit, 2015-09-08:
btw, here is a tipp: toggle off keyframe markers and this error will not happen ;) best solution for me so far. you can toggle it on after KF has reached 1024 or before. however if KF markers are on and the 1024th KF will be added, rviz will crash.