raulmur/ORB_SLAM #22: camera poses of all the frames

vibhav-vineet, 2015-04-30:
Hi Raul,

Thanks for sharing the code.
I want to save the camera pose of all the frames, not only for the key frames. Could you please tell how to do this?

thanks,

raulmur, 2015-05-04:
The Frame class stores the pose in mTcw which is a cv::Mat 4x4 of type float (CV_32F)
It transforms points in the world coordinate system into the camera.
You could access this member at the end of GrabImage function in Tracking. However you should consider that frames are not optimized with bundle adjustment, being potentially inconsistent with the keyframes. For example when the map contains few keyframes, and all of them are included in the local bundle adjustment, the global scale of the reconstruction might change and frame poses will not notice.

geokarant, 2016-10-17:
hello Raul, thank you a lot for sharing the code... Can you please tell me where in the code i can find where you set the origin of the world coordinate system and how it is transformed into the camera one?

thank you very much :)