raulmur/ORB_SLAM #25: scale changed after turnning

liqile, 2015-05-04:
hi, @raulmur 
thanks for your sharing your orb-slam code .sorry to trouble you but I have a trouble while tracking
I firstly go straight towards north and then turn to east
orb-slam can calculate the degrees very well, but the scale of distance changed obviously after turning to east.
could you give me some suggestions ?
thanks and best

liqile, 2015-05-04:
I know what's the reason now.
orbslam will use the velocity of last 2 frames, this influence may be remained after changing a direction, so I just remove the utilizing of velocity, and it works well.

Gingol, 2015-05-06:
Hi @liqile 
I notice the same problem, can you tell me what kind of changes you've done?

liqile, 2015-05-06:
hi @Gingol 
in the source file `"Tracking.cc"`, you can find the function 
`void Tracking::GrabImage(const sensor_msgs::ImageConstPtr& msg)`
on the line 229~236, you can find following code:
`
            if(!mbMotionModel || mpMap->KeyFramesInMap()<4 || mVelocity.empty() || mCurrentFrame.mnId<mnLastRelocFrameId+2)
                bOK = TrackPreviousFrame();
            else
            {
                bOK = TrackWithMotionModel();
                if(!bOK)
                    bOK = TrackPreviousFrame();
            }
`
what I have done is that to change the code above to following:
`
bOK=TrackingPreviousFrame();
`

liqile, 2015-05-06:
@Gingol 
you can try to combine orb slam with imu, this may also deal with that problem

liqile, 2015-05-06:
@Gingol do you have QQ？

Gingol, 2015-05-07:
@liqile 
1) I've tried to change the code like you said but it doesn't work. After a turning the distance between the poses of the camera in rviz are very small or in some cases it seems that the camera doesn't move at all.
2) In our project we want to use only the camera.
3) No I haven't QQ.

liqile, 2015-05-07:
@Gingol what's the number of features in your proj.? I set the feature amount to 8000

Gingol, 2015-05-07:
@liqile I kept the default number 1000.
Can you tell me which camera and which lens you are using?

liqile, 2015-05-08:
@Gingol 
720 x 1280

liqile, 2015-05-08:
@Gingol 
can you give me your ros bag with the camera calibration?
li_qi_le@163.com

liqile, 2015-05-09:
@Gingol 
I think you can use the camera on your iphone

Oblynx, 2015-05-09:
Hello @liqile,
You mentioned combining Orbslam with an IMU. How do you propose to implement that, perhaps with a Kalman filter?

versatran01, 2015-05-13:
@Oblynx the easiest the approach would be to replace the constant velocity motion model with imu motion model.

mhkabir, 2015-05-17:
@versatran01 Can you elaborate on how we can do that?
Would metric scale estimation be possible with such a approach?

shabhu18, 2015-05-17:
I am also facing the same problem . Has anyone come up with a solution .I donot have the IMU in my case.

liqile, 2015-05-19:
@shabhu18 
hi, I think you can take utilization of a new camera. I used the camera on a xiaomi phone , and it worked well

versatran01, 2015-05-19:
@mhkabir To start with, you could just integrate the inter-frame gyro readings as the guess of inter-frame rotation. The accelerometer readings are probably too noisy to estimate translation.

haoanw, 2015-10-11:
@liqile hello, about your comment here:
#######
"in the source file "Tracking.cc", you can find the function 
void Tracking::GrabImage(const sensor_msgs::ImageConstPtr& msg)
on the line 229~236, you can find following code:

229: if(!mbMotionModel || mpMap->KeyFramesInMap()<4 || mVelocity.empty() || mCurrentFrame.mnId<mnLastRelocFrameId+2)
230: bOK = TrackPreviousFrame();
231: else
232: {
233: bOK = TrackWithMotionModel();
234: if(!bOK)
235: bOK = TrackPreviousFrame();
236: }
what I have done is that to change the code above to following:
 bOK=TrackingPreviousFrame(); "
#######

how did you observe this line was in charge of the model velocity? and how did this velocity been determined? could you please give me some hints? i still could't find out how the scale of map been calculated. thx

AlexandreBorowczyk, 2016-01-26:
Hi @mhkabir, did you end up looking in to changing the velocity motion model? I've experimented the combinaition of ORB_SLAM with MSF but the results are still inconclusive.

mhkabir, 2016-01-26:
@AlexandreBorowczyk No, I gave up on it. Rolled our own visual-inertial odometry system.