raulmur/ORB_SLAM #32: Tracking on Feature-Poor Environments: Alternate Feature Extractors/Descriptors

Oblynx, 2015-05-28:
Hello,
It would be interesting to combine this SLAM system with a feature extractor and descriptor *other than __ORB__*, primarily to compare performance in special use cases, such as *environments with few corners*. However the ORB version implemented here is greatly extended with specific functionality which probably wouldn't be easy to convert.
**- Does anyone have an idea on this subject?**

jonas-, 2015-12-08:
Hi
I'm starting to look into this topic with using line features in addition to the existing ORB-SLAM-Framework (which I adaptded for a catadioptric omnicam) for more robust tracking, mapping.

Did you @Oblynx or somebody else already incorporated additional features?
Would be great if you could share your experience, also with the implementation.

Cheers, Jonas

Oblynx, 2015-12-09:
Hi Jonas,
No, I haven't after all. I did experiment with different configuration
settings for ORB-SLAM, such as FAST threshold, however what made the
biggest impact for me by far was simply increasing the light level in the
room. It's been quite some time so I don't remember any other details.
On a tangent, this paper might be interesting:
http://web.stanford.edu/class/cs231m/projects/final-report-shridhar-neo.pdf

*-------------------------------------------------------*
*  Konstantinos Samaras-Tsakiris*
School of Electrical and Computer Engineering
Aristotle University of Thessaloniki

nlw0, 2015-12-09:
@jonas- That is a topic I also have some interested on, would love to see any work you have done in edge-based SLAM.

jonas-, 2015-12-10:
I haven't done any work with line, edge features yet. But start now and will share my results as soon as I have some.