raulmur/ORB_SLAM #34: how can i run this on ROS(jade)?

ashclission, 2015-06-11:
when i install ORB_SLAM , i have some problem.
/******************************************************************************
Failed to invoke /opt/ros/jade/bin/rospack deps-manifests ORB_SLAM-master
[rospack] Error: package 'ORB_SLAM-master' depends on non-existent package 'opencv2' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
******************************************************************************/
how can i solve this problem . Thanks.

MarcGyongyosi, 2015-06-11:
Hi, you need to take out all the opencv stuff from the depends in the package.xml - I had that error pop up, too, despite opencv2 being installed from ROS Hydro core

Best,
Marc

ashclission, 2015-06-12:
Thanks Marc. it does work .and  i going to run this on my android robot.

gitunit, 2015-06-13:
@ashclission how do you run it on android? have you installed ros on it or do you only use the images from the smartphone?

ashclission, 2015-06-15:
just for a test. i see a ndk version on android. and going to install it on
my phone.

ashclission, 2015-06-15:
is it  *possible**?*

2015-06-15 9:12 GMT+08:00 wj mickey <bonwind99@gmail.com>:

jwcrawley, 2015-07-07:
Just because I did not see a confirmation regarding ORB-SLAM and ROS-Jade, I do confirm it works on a stock Ubuntu 14.04, using a Microsoft Surface 1 tablet. And it works rather well.

@ashclission I know that I've seen LSD-slam work on an Android phone in a visual odometry mode.

wangchuan, 2015-09-21:
@jwcrawley I tried to build ORB-SLAM with ROS-Jade in Ubuntu 14.10, but encountered a problem issued at https://github.com/raulmur/ORB_SLAM/issues/65. Could you help me analyze what may cause the problem? Thanks!

nicholas-tien, 2016-12-29:
@MarcGyongyosi I don't know how to  take out all the opencv stuff from the depends in the package.xml?Where to find the package.xml and how to remove opencv thing? Can you help me

bdjohnson529, 2018-01-31:
"package.xml" has been renamed to "manifest.xml".

Removing the opencv2 dependency worked on Ubuntu 16.04.

ravikt, 2018-06-12:
I am getting this error after removing opencv from xml file 
fatal error: cv_bridge/cv_bridge.h: No such file or directory
Do I have to change the code in ros_rgbd.cc