raulmur/ORB_SLAM #36: ORB_SLAM/Frame, Map ,Camera, World

5aou, 2015-06-15:
Hi, 

can you show me where I can find Frame , Map,Camera and World because I don't find them under ORB_SLAM (in general I think that I didn't understand those two parts : 
2-The last processed frame is published to the topic /ORB_SLAM/Frame. You can visualize it using image_view and number 3).

Thank you a lot.

gitunit, 2015-06-16:
it seems you need to learn the basics of ROS first. have a look here: http://wiki.ros.org/ROS/Tutorials

5aou, 2015-06-16:
Thank you a lot for your response, it's exactly what I'm doing, thank you again.

5aou, 2015-06-22:
Hi again gitunit,

I manage to understand what that means, instead I've encounter the same issue that you did when trying "rosrun image_view image_view image:=/ORB_SLAM/Frame _autosize:=true", I searched in the internet and I found that you resolved that problem by downgrading the image_pipeline, but I don't know how to do that (I searched but I don't find the good package),can you help me please (I'm on ubuntu 14.04 , ROS Indigo).
Tank you a lot.

gitunit, 2015-06-22:
as far as i remember i grabed version 1.12.10 from [here](https://github.com/ros-perception/image_pipeline/releases) and put it into catkin workspace src folder. then you run catkin_make (from one level up). don't forget to source your workspace. that should be it, i guess.

5aou, 2015-06-23:
thank you for your reply, unfortunately it did not work, I have not a catkin workspace src folder but I created one, even if I did what you told me it didn't work, any other suggestions  please.

PS: I don't understand this sentence "don't forget to source your workspace".

Tank you a lot.

gitunit, 2015-06-23:
what exactly did not work? did you setup your workspace like described [here](http://wiki.ros.org/ROS/Tutorials/InstallingandConfiguringROSEnvironment) ?

5aou, 2015-06-24:
yes I did, beside that I started all from zero and I encounter a new error when I do : "cmake .. -DROS_BUILD_TYPE=Release" (That I had not in the first time), The error is : 
 ORBextractor.cc:606:63: error: ‘FAST’ was not declared in this scope
                 FAST(cellImage,cellKeyPoints[i][j],fastTh,true);
                                                               ^
/home/khaoula/ORB_SLAM/src/ORBextractor.cc:615:34: error: ‘ORB’ has not been declared
                 if( scoreType == ORB::HARRIS_SCORE )
                                  ^
/home/khaoula/ORB_SLAM/src/ORBextractor.cc:682:17: error: ‘KeyPointsFilter’ has not been declared
                 KeyPointsFilter::retainBest(keysCell,nToRetain[i][j]);
                 ^
.......................

(I think that I have to add a header that I don't know),really stranded here.

Thank you.

5aou, 2015-06-25:
re, I resolved this error , now when I do rosrun ORB_SLAM ORB_SLAM...... I get this one : Erreur de segmentation (core dumped)
any help please, and thank you.

chaiein, 2015-11-03:
even I have the same problem as 5aou can please tell me how to solve it