raulmur/ORB_SLAM #37: Error while building ORB_SLAM

sj312931, 2015-06-18:
Hello,

I am trying to build ORB_SLAM package. I am using ROS-Indigo on ubuntu 14.04. I get the following error on step 6 (https://github.com/raulmur/ORB_SLAM). 

~/ORB_SLAM/build$ cmake .. -DROS_BUILD_TYPE=Release
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
[rosbuild] Building package ORB_SLAM
Failed to invoke /opt/ros/indigo/bin/rospack deps-manifests ORB_SLAM
[rospack] Error: package 'ORB_SLAM' depends on non-existent package 'opencv2' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'

CMake Error at /opt/ros/indigo/share/ros/core/rosbuild/public.cmake:129 (message):

 Failed to invoke rospack to get compile flags for package 'ORB_SLAM'.  Look above for errors from rospack itself.  Aborting.  Please fix the broken dependency!

Call Stack (most recent call first): /opt/ros/indigo/share/ros/core/rosbuild/public.cmake:207 (rosbuild_invoke_rospack)
CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/anarang/ORB_SLAM/build/CMakeFiles/CMakeOutput.log".

Can somebody please help me out? I am new to ROS.

MarcGyongyosi, 2015-06-18:
take out all opencv dependencies from the package.xml.... that does the trick. Since ROS Hyrdo, opencv is included in core

sj312931, 2015-06-18:
Thank you for the quick response Mark. I am able to do cmake without getting any errors but when I go onto the next step and do make I get the following errors:

~/ORB_SLAM/build$ make
[rosbuild] Building package ORB_SLAM
[rosbuild] Cached build flags older than manifests; calling rospack to get flags
-- Using CATKIN_DEVEL_PREFIX: /home/anarang/ORB_SLAM/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() for dry packages
-- Using CATKIN_TEST_RESULTS_DIR: /home/anarang/ORB_SLAM/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.14
[rosbuild] using multiarch 'x86_64-linux-gnu' for finding Boost
-- Using these message generators: gencpp;genlisp;genpy
[rosbuild] Including /opt/ros/indigo/share/roslisp/rosbuild/roslisp.cmake
[rosbuild] Including /opt/ros/indigo/share/roscpp/rosbuild/roscpp.cmake
[rosbuild] Including /opt/ros/indigo/share/rospy/rosbuild/rospy.cmake
Build type: Release
-- Configuring done
-- Generating done
-- Build files have been written to: /home/anarang/ORB_SLAM/build
[  0%] Built target rospack_genmsg_libexe
[  0%] Built target rosbuild_precompile
[  0%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/main.cc.o
In file included from /home/anarang/ORB_SLAM/include/ORBVocabulary.h:26:0,
                 from /home/anarang/ORB_SLAM/include/KeyFrame.h:28,
                 from /home/anarang/ORB_SLAM/include/MapPoint.h:25,
                 from /home/anarang/ORB_SLAM/include/FramePublisher.h:25,
                 from /home/anarang/ORB_SLAM/include/Tracking.h:29,
                 from /home/anarang/ORB_SLAM/src/main.cc:29:
/home/anarang/ORB_SLAM/Thirdparty/DBoW2/DBoW2/TemplatedVocabulary.h: In instantiation of ‘DBoW2::TemplatedVocabulary<TDescriptor, F>::~TemplatedVocabulary() [with TDescriptor = cv::Mat; F = DBoW2::FORB]’:
/home/anarang/ORB_SLAM/src/main.cc:86:29:   required from here
/home/anarang/ORB_SLAM/Thirdparty/DBoW2/DBoW2/TemplatedVocabulary.h:510:3: warning: deleting object of abstract class type ‘DBoW2::GeneralScoring’ which has non-virtual destructor will cause undefined behaviour [-Wdelete-non-virtual-dtor]
   delete m_scoring_object;
   ^
/home/anarang/ORB_SLAM/Thirdparty/DBoW2/DBoW2/TemplatedVocabulary.h: In instantiation of ‘void DBoW2::TemplatedVocabulary<TDescriptor, F>::createScoringObject() [with TDescriptor = cv::Mat; F = DBoW2::FORB]’:
/home/anarang/ORB_SLAM/Thirdparty/DBoW2/DBoW2/TemplatedVocabulary.h:420:23:   required from ‘DBoW2::TemplatedVocabulary<TDescriptor, F>::TemplatedVocabulary(int, int, DBoW2::WeightingType, DBoW2::ScoringType) [with TDescriptor = cv::Mat; F = DBoW2::FORB]’
/home/anarang/ORB_SLAM/src/main.cc:86:29:   required from here
/home/anarang/ORB_SLAM/Thirdparty/DBoW2/DBoW2/TemplatedVocabulary.h:446:3: warning: deleting object of abstract class type ‘DBoW2::GeneralScoring’ which has non-virtual destructor will cause undefined behaviour [-Wdelete-non-virtual-dtor]
   delete m_scoring_object;
   ^
[  0%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/Tracking.cc.o
[  0%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/LocalMapping.cc.o
[  0%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/LoopClosing.cc.o
[  0%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/ORBextractor.cc.o
[  0%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/ORBmatcher.cc.o
[  0%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/FramePublisher.cc.o
[  0%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/Converter.cc.o
[  0%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/MapPoint.cc.o
[  0%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/KeyFrame.cc.o
[  0%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/Map.cc.o
[  0%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/MapPublisher.cc.o
[  0%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/Optimizer.cc.o
[  0%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/PnPsolver.cc.o
[  0%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/Frame.cc.o
[  0%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/KeyFrameDatabase.cc.o
[  0%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/Sim3Solver.cc.o
[  0%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/Initializer.cc.o
Linking CXX executable ../bin/ORB_SLAM
/usr/bin/ld: cannot find -l:/usr/lib/x86_64-linux-gnu/libtinyxml.so
collect2: error: ld returned 1 exit status
make[2]: *** [../bin/ORB_SLAM] Error 1
make[1]: *** [CMakeFiles/ORB_SLAM.dir/all] Error 2
make: *** [all] Error 2

Do have an idea how to fix this?

closed 2015-06-22 by sj312931

tahmine-iaun, 2016-09-06:
Hi! well i faced to similar issue but to my suprise i dont receive any errs just :
[image]
](url)
i donno why it just remain pending no err no results! anyone can help?