raulmur/ORB_SLAM #38: ORB_SLAM initialization failure on indigo ununto 14.04

iamfarhan007, 2015-06-19:
Hello 
i have remapped the camera topics correctly . but getiing this error 
RB-SLAM Copyright (C) 2014 Raul Mur-Artal
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Loading ORB Vocabulary. This could take a while.

Wrong path to vocabulary. Path must be absolut or relative to ORB_SLAM package directory.
[orb_slam-3] process has died [pid 30997, exit code 1, cmd /home/farhan/catkin_ws/devel/lib/orb_slam/orb_slam Data/ORBvoc.yml Data/Settings.yaml /camera/image_raw:=/creative_cam/image_color __name:=orb_slam __log:=/home/farhan/.ros/log/2e26604a-16ae-11e5-9789-d8fc9361bda4/orb_slam-3.log].
log file: /home/farhan/.ros/log/2e26604a-16ae-11e5-9789-d8fc9361bda4/orb_slam-3*.log

Thanks 
Farhan

dlime, 2015-06-30:
Did you uncompressed Data/ORBvoc.yml.tar.gz ?

iamfarhan007, 2015-07-09:
Thanks I decompressed it now . ot builds fine

mihaibujanca, 2015-08-21:
@iamfarhan007 Please close the issue if it's all ok now

iamfarhan007, 2015-08-21:
Thanks I works fine now.

closed 2015-08-21 by iamfarhan007