raulmur/ORB_SLAM #39: Different error buildiing ORB-SLAM

jwcrawley, 2015-06-21:
Installation: Ubuntu 14.04
ROS: Indigo
Output of $ROS_PACKAGE_PATH=/home/josh/Desktop/ORB-SLAM:/opt/ros/indigo/share:/opt/ros/indigo/stacks

Error occurs in building ORB-SLAM at step 3-6. Output of error:

josh@josh-Surface-with-Windows-8-Pro:~/Desktop/ORB_SLAM/build$ cmake .. -DROS_BUILD_TYPE=RELEASE
[rosbuild] Building package ORB_SLAM
[rosbuild] Error from directory check: /opt/ros/indigo/share/ros/core/rosbuild/bin/check_same_directories.py  /home/josh/Desktop/ORB_SLAM
1
Traceback (most recent call last):
  File "/opt/ros/indigo/share/ros/core/rosbuild/bin/check_same_directories.py", line 46, in <module>
    raise Exception
Exception
CMake Error at /opt/ros/indigo/share/ros/core/rosbuild/private.cmake:102 (message):
  [rosbuild] rospack found package "ORB_SLAM" at "", but the current
  directory is "/home/josh/Desktop/ORB_SLAM".  You should double-check your
  ROS_PACKAGE_PATH to ensure that packages are found in the correct
  precedence order.
Call Stack (most recent call first):
  /opt/ros/indigo/share/ros/core/rosbuild/public.cmake:177 (_rosbuild_check_package_location)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/josh/Desktop/ORB_SLAM/build/CMakeFiles/CMakeOutput.log".

Further research on answers.ros.org indicates that using CMAKE isn't supported, given the error above. (source: http://answers.ros.org/question/65801/ros-inside-part-of-a-c-project/ )

Does anyone have any sort of answers how to make this work? This would be an impressive project to get up and running. 

Sincerely,
Josh Conway

sj312931, 2015-06-22:
Hi jwcrawley,

You need add the ROS_PACKAGE_PATH for your ORB_SLAM package in the bashrc. Please check if you have added it correctly.

Cheers

jwcrawley, 2015-06-22:
Indeed, I added the appropriate export for ROS_PACKAGE_PATH and also verified the compile directory was included in its path. And it still gives the error. I'll try in /opt/ros/indigo/share , but as of now, it still does not work.

jwcrawley, 2015-06-22:
For some reason, some of my source tree was incomplete. rm'ing and redownloading from source git solved issue.

Now, it's the well known opencv2 issue, which a solution is already stated. 

Thank you,
Josh Conway

closed 2015-06-22 by jwcrawley

riaverma96, 2015-07-16:
Hi jwcrawley,
I am getting the very same error as you. 
My ROS_PACKAGE_PATH cannot find the ORB_SLAM package. I am using ~/.bashrc to export ROS_PACKAGE_PATH and include different directories for ROS to look into. but when I tried to find packages called ORB_SLAM in my computer, it says "No packages found"

Is this a OpenCV issue? Can you explain how you solved it in a little more detail?
Thanks,
Ria

jwcrawley, 2015-07-16:
Greetings @riaverma96 ,

That was an annoying error I kept getting, but I think I've solved it now. I'll start from the beginning, which seems to have worked on 2 separate machines:

(Be aware, I only recommend the 1st step for completely clean install)
1. Use Synaptic and completely remove ALL ros packages
2. Follow steps on http://www.ros.org/install/ for Desktop-Full Install. I've used Jade. 
3. Go into your ~/.bashrc and remove all instances of ROS (should be a source and an export line)
4. At the bottom of the ~/.bashrc, you want to put the following:
    source /opt/ros/jade/setup.bash
    export ROS_PACKAGE_PATH=~/catkin_ws:$ROS_PACKAGE_PATH
5. Get ORB_SLAM into your ~/catkin_ws directory va git. After this is done, you'll have ~/catkin_ws/ORB_SLAM/{stuff}
6. Edit ORB_SLAM/manifest.xml and remove the whole line that mentions opencv2
7. Follow the compile directions appropriately for 3rd party and then main.
8. Add a node to the ExampleGroovyHydro.launch at the top (beneath the <launch> )
    <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" output="screen" >
        <param name="video_device" value="/dev/video0" />
        <param name="image_width" value="640" />
        <param name="image_height" value="480" />
        <param name="pixel_format" value="mjpeg" />
        <param name="camera_frame_id" value="usb_cam" />
        <param name="io_method" value="mmap"/>
    </node>
9. Run roscore in a terminal.
10. Now use roslaunch to launch that file. It should spin up rviz and a webcam viewer.
Should be up and running.

riaverma96, 2015-07-21:
Hi jwcrawley,

Thanks :) I followed your steps and it worked out great! I am really curious as to why we are putting the ORB-SLAM package inside catkin_ws instead of its src file though. Is this what I should do in the future with packages?

Thanks for all your help,
Ria

wangchuan, 2015-09-23:
@jwcrawley I added the web cam node as you provided but there is still no frame capturing (black screen at the image_viewer window). But I tested the camera driver is Okay. So what's the correct way to connect the ORB SLAM with a real USB camera to do online testing?

pmoulon, 2015-11-01:
In my case I installed sudo apt-get install ros-hydro-usb-cam since the message was appearing ERROR: cannot launch node of type [usb_cam/usb_cam_node]: usb_cam.

@jwcrawley I followed your step but the image viewer remains black and the mapping process does not start.
I know that my webcam is working inside ROS since I have tried the following launcher
 ```
<launch>
  <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video0" />
    <param name="image_width" value="320" />
    <param name="image_height" value="240" />
    <param name="pixel_format" value="mjpeg" />
    <param name="camera_frame_id" value="usb_cam" />
    <param name="io_method" value="mmap"/>
    <param name="framerate" value="30"/>
  </node>
  <node name="image_view" pkg="image_view" type="image_view" respawn="false" output="screen">
    <remap from="image" to="/usb_cam/image_raw"/>
    <param name="autosize" value="true" />
  </node>
</launch>
```

extr15, 2016-06-16:
first, check if it is done: export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:your_path_to_ORB_SLAM
second, look into the error info:

> CMake Error at /opt/ros/indigo/share/ros/core/rosbuild/private.cmake:102 (message):

> [rosbuild] rospack found package "ORB_SLAM" at "", but the current

private.cmake, line 102. it seems that ${pkgname}_PACKAGE_PATH is null, this variable is assigned in  the macro "rosbuild_find_ros_package" in public.cmake. thus you can check if the command "rospack find ${pkgname}" is OK. 

> rospack find ORB_SLAM.

in my case, the output is 
> rospack: error while loading shared libraries: libboost_filesystem.so.1.54.0: cannot open shared object file: No such file or directory

in conclusion, it may be caused by the wrong link version lib.

KathiravanNatarajan, 2017-02-04:
@jwcrawley : can I implement it in ARDRONE instead of webcam ? Should I have to make any changes in the launch file ??

archenroot, 2017-06-13:
@jwcrawley - could you post the real value of your ROS_PACKAGE_PATH? I am using catkin_make workspace and trying to make this works...
Thx.