raulmur/ORB_SLAM #41: How to visualize the map with color?

Po-Jen, 2015-06-26:
Hello, in the video below:
https://www.youtube.com/watch?t=30&v=HlBmq70LKrQ

There are two versions of map, the first is the one contains dots in red and black
http://imgur.com/ra566zO

And another version is colorful
http://imgur.com/xn2RG5o

I've tried to find possible topic publishing the colorful version, but I haven't found one. (I've seen /ORB_SLAM/Map_array, but it's a empty topic) Also, I've checked ROS parameters and found nothing related. So I can only get the first version in my Rviz.
http://imgur.com/UDXhtgO

Should I modify the code or do something else to see the colorful version?
Thanks in advance.

NikolausDemmel, 2015-06-26:
The  video shows semi dense mapping, which I guess is not yet released as source code yet.

This repo so far seems to only contain the sparse SLAM part.

Po-Jen, 2015-06-27:
@NikolausDemmel Thanks. No wonder I cannot find the way to show semi dense map in Rviz.

closed 2015-06-27 by Po-Jen