raulmur/ORB_SLAM #42: rosrun image_view

5aou, 2015-07-02:
Hi all,
I had a black window when I did:
rosrun image_view image_view image:=/ORB_SLAM/Frame _autosize:=true, any suggestion please.
PS: I'm using a Logitech camera.
Thank you a lot.

Po-Jen, 2015-07-03:
There's no need to view image in image_view, I also have black window but I can see the map in Rviz.

Make sure your image is published to /camera/image_raw, then move your camera so that map can be initialized and viewed in Rviz.

5aou, 2015-07-06:
Hi again,thank you a lot for your response,apparently my image exists : 
rostopic list
/ORB_SLAM/Frame
/ORB_SLAM/Map
/camera/image_raw
/clicked_point
/initialpose
/move_base_simple/goal
/rosout
/rosout_agg
/tf
/tf_static
but when I moved the camera I have no results,I have this :  
[image]
Thank you again.

Po-Jen, 2015-07-06:
Can you check if /camera/image_raw correctly shows the image in Rviz?

5aou, 2015-07-06:
How to do this please ?

Po-Jen, 2015-07-06:
Add a new display, and choose Image. After that, choose /camera/image_raw to be the topic showed.
http://wiki.ros.org/rviz/UserGuide#Adding_a_new_display

5aou, 2015-07-06:
I don't find the topic /camera/image_raw !!

[image]
How can I solve this ? can this be caused by the camera ? (I'm using a logitech)

Po-Jen, 2015-07-06:
In the image you just posted, you can see that under Image, there's a row called "Image Topic" (right above the "Transport Hint"). Click on the blank space above "raw" and choose /camera/image_raw.

5aou, 2015-07-07:
I didn't find it , I have just /ORB_SLAM/Frame as a choice !

Po-Jen, 2015-07-07:
OK, that's the problem. You should check why your camera driver does not publish image to /camera/image_raw. If there's no image published to this topic, ORB_SLAM as a subscriber, cannot build map since there's no input.

This should not be hard, leverage ROS answer.
http://answers.ros.org/questions/

5aou, 2015-07-07:
OK,thank you a lot for you help, I'll post my issue there then.

5aou, 2015-07-07:
re, why it's shown here : rostopic list
/ORB_SLAM/Frame
/ORB_SLAM/Map
/camera/image_raw
/clicked_point
/initialpose
/move_base_simple/goal
/rosout
/rosout_agg
/tf
/tf_static
??

Po-Jen, 2015-07-07:
It's strange to me too. I saw you have /camera/image_raw in rostopic list, so I let you to check the content in Rviz. 

Maybe you can use rqt to see which node publish this topic, and use `rostopic echo /camera/image_raw` to check the content.

5aou, 2015-07-08:
this is what I get : 
[image]

Po-Jen, 2015-07-08:
I suggest yo to learn some basic ROS usage, you didn't change the setting in rqt to show all topics.  I said "use rqt to see which node publish this topic", but since you didn't show topics, we'll not get which node publish /camera/image_raw.

And BTW, you didn't even start a driver node, how can ORB_SLAM get image?

5aou, 2015-07-08:
I don't know how to do that, I'm really lost between the lines of ros tutorials....

5aou, 2015-07-08:
I forget to tell you that I changed the settings in rqt but the results don't change.... Can you tell me please how can I start a driver node, I searched / net but I didn't find how to do it...

jwcrawley, 2015-07-08:
You need a camera to output frames into the input for ORB-SLAM.

I used usb_cam module to get the camera data. However, usb_cam sends camera data to /usb_cam/image_raw and we need it to put data into /camera/image_raw

to see if your camera works, only need to run in one terminal:
> rosrun usb_cam usb_cam_node

And the viewer in another terminal:
> rosrun image_viewer image_viewer image:=/usb_cam/image_raw

If the above doesn't work, you have a camera that doesn't work properly. Check it in Cheese before continuing. Check the video device and the available resolutions.

Now, for changing resolutions for getting a better image, it's simpler to make a roslaunch file (go read $ROS_PACKAGE_DIR/ORB_SLAM/ExampleGroovyHydro.launch ) and **add** in the following for the first node:

> <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" output="screen" >
>   <param name="video_device" value="/dev/video0" />
>   <param name="image_width" value="640" />
>    <param name="image_height" value="480" />
>    <param name="pixel_format" value="yuyv" />
>    <param name="camera_frame_id" value="camera" />
>    <param name="io_method" value="mmap"/>
>    <remap from="/usb_cam/image_raw" to="/camera/image_raw" />
>  </node>

 You will have to modify the appropriate settings to set it up for your system. If you have multiple cameras, change /dev/ appropriately.

Now run the launch file by 'roslaunch ExampleGroovyHydro.launch' . It should spin up the camera, xViz as well as a video window that first reports something like 'loading definitions'. It should load, and then show you video soon after. You're up. 

Now, after playing with it, and investigating the settings, I also made a change in Data/Settings.yaml . I changed "UseMotionModel" to 0 . I'm still investigating the other settings, and how they affect the result.

5aou, 2015-07-08:
you saved my life, thank you a lot , it  works. But I have always this error : Fixed Frame [map] does not exist, do you encounter the same error ? I don't know what is the cause !

jwcrawley, 2015-07-08:
Yes. You'll get that until you start generating the map. 

You'll see a b/w image of your webcam with green lines. Start moving around, usually by walking forward and back. Don't start by turning in place. Once you do enough frames, it'll then switch from lines to little squares showing keypoint mappings. That's when you're starting to build the 3d map for the SLAM. 

And that error you noted will go away.

5aou, 2015-07-09:
thank you a lot for your explanation, how much time it takes to get the frames ?,because it doesn't work for me...!!!
PS : I don't see a b/w image of the webcam with green lines.
[4](https://cloud.githubusercontent.com/assets/12268902/8592061/ba3119a8-262c-11e5-8df8-552557985b56.png)

5aou, 2015-07-10:
re, can you tell me please how do you chose "Fixed Frame" in Global Options/ RViz, because I noticed that sometimes I have "map" and others I have "ORB_SLAM/Camera" !! thing that I find weird
PS: Can you please give me some easy tutorials on ROS,that are related to all that,because I'm new in SLAM and ROS.
Thank you a lot.

bjornph, 2015-07-24:
See the author's youtube channel for some successful datasets: https://www.youtube.com/channel/UC_jh6pkpTd3movgveqdILuQ

I would suggest you try to get the example sequence to run first:
https://github.com/raulmur/ORB_SLAM#5-example-sequence

Once you get this working, make a copy of the file Data/Settings.yaml and use your own camera calibration settings here. fx,fy,cx,cy and lens distortion parameters. To launch this settings file, make a copy of the orb slam launch file under launch/ folder. Here you have to so you point to your own settings file and also remap fix remap: <remap from="/camera/image_raw" to="YOUR IMAGE TOPIC" /> if you use the usb_cam as jwcrawley suggested, then it will  be <remap from="/camera/image_raw" to="/usb_cam/image_raw />

------
If you are serious about using ORB_SLAM and/or other algorithms for SLAM purpose then I recommend you to learn the  basics of both SLAM and ROS. Depending on your background this might take some time, but well worth investing time as otherwise you'll have a bad time.

SLAM:
http://wiki.ros.org/ROS/Tutorials
Computer Vision/SLAM
This is a nice introduction to history of monocular SLAM, at least the first 25 mins are well worth watching: http://videolectures.net/bmvc2012_davison_scene_perception/
Visual SLAM introduction slides:
http://frc.ri.cmu.edu/~kaess/vslam_cvpr14/
If you are completely new to Computer Vision, please see:
http://szeliski.org/Book/

And an introduction to CV for this purpose, such as Multiple view geometry lessons:
https://www.youtube.com/watch?v=RDkwklFGMfo&list=PLTBdjV_4f-EJn6udZ34tht9EVIW7lbeo4&index=1

5aou, 2015-07-29:
Hi, thank you a lot for your precious help, the example works well,I did all the changes but it didn't work.
PS: I don't understand this part of your answer "To launch this settings file, make a copy of the orb slam launch file under launch/ folder. Here you have to so you point to your own settings file and also remap fix remap: if you use the usb_cam as jwcrawley suggested, then it will be http://wiki.ros.org/ROS/Tutorials
Computer Vision/SLAM".

ldhl103, 2015-09-08:
I think I met the same prolem as 5aou met....

[image]

I can use usb cam open my cam , and roslaunch ExampleGroovyHydro.launch 
But I don't see any map when I move my camera.

I think it seems that I don't publish to camera/image_raw successfully......

how do I solve this problem?

I will appreciate the help
Thanks!

mairasaboia, 2015-11-23:
I have a similar issue. I can see the b/w image of my webcam with green lines but they never change to little squares as aforementioned by @jwcrawley . I have walked back and forth with the camera in a highly illuminated environment but the method refused to initialize as one can see in the image below.

[image]

My camera is a PlayStation eye camera. The data that I get from opencv calibration is:
image_width: 640
image_height: 480
camera_name: narrow_stereo
camera_matrix:
  rows: 3
  cols: 3
  data: [688.701702, 0, 364.134285, 0, 692.843542, 158.226495, 0, 0, 1]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.027262, -0.168572, -0.013315, -0.001805, 0]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1, 0, 0, 0, 1, 0, 0, 0, 1]
projection_matrix:
  rows: 3
  cols: 4
  data: [677.89978, 0, 364.943701, 0, 0, 676.806702, 153.303095, 0, 0, 0, 1, 0]

Do I need to use some rectification mechanism?

Can someone help me going through this problem?

Thanks, mss

mhzaki, 2015-12-29:
Hello!
did you resolve this problem because i have the same one and i don' know how fix it 
please help me.

mhzaki, 2015-12-29:
Hello!
did you resolve this problem ? because i have the same one and i don't know how fix it ..!!
please help me.

5aou, 2015-12-29:
Hi, unfortunately I didn't resolve it. Sorry.

mhzaki, 2015-12-31:
Hi 
Ok sorry and thank you ! if i find a solution I will tel you ... 
bye.

5aou, 2016-01-01:
Ok that'll be noce.

corfox, 2016-04-11:
@jwcrawley This work for me thanks to your suggestion!

qinmengsu, 2016-07-04:
@5aou @ldhl103 I think I have the same problem with guys above, images can be shown in rviz by following @jwcrawley launch file, but still no map display, I use uvc_camera, don't know what's problem happened, rqt_graph shows no rosmsg send from uvc_camera/image_raw to camera/image_raw. Who can help us?

tahmine-iaun, 2016-12-25:
@jwcrawley  
hi what kind of usb cam module you used?

tahmine-iaun, 2016-12-25:
[image]
@jwcrawley    i use a webcam to get images but i got this msg