raulmur/ORB_SLAM #44: Error during make Indigo

husdo, 2015-07-06:
Hi,
I've got that error message during the ORB_SLAM make. I've done everything like in the manual.

Regards,
husdo

 /opt/ros/indigo/share/ORB_SLAM/Thirdparty/DBoW2/DBoW2/TemplatedVocabulary.h: In instantiation of ‘DBoW2::TemplatedVocabulary<TDescriptor, F>::~TemplatedVocabulary() [with TDescriptor = cv::Mat; F = DBoW2::FORB]’:
/opt/ros/indigo/share/ORB_SLAM/src/main.cc:86:29:   required from here
/opt/ros/indigo/share/ORB_SLAM/Thirdparty/DBoW2/DBoW2/TemplatedVocabulary.h:510:3: warning: deleting object of abstract class type ‘DBoW2::GeneralScoring’ which has non-virtual destructor will cause undefined behaviour [-Wdelete-non-virtual-dtor]
   delete m_scoring_object;
   ^
/opt/ros/indigo/share/ORB_SLAM/Thirdparty/DBoW2/DBoW2/TemplatedVocabulary.h: In instantiation of ‘void DBoW2::TemplatedVocabulary<TDescriptor, F>::createScoringObject() [with TDescriptor = cv::Mat; F = DBoW2::FORB]’:
/opt/ros/indigo/share/ORB_SLAM/Thirdparty/DBoW2/DBoW2/TemplatedVocabulary.h:420:23:   required from ‘DBoW2::TemplatedVocabulary<TDescriptor, F>::TemplatedVocabulary(int, int, DBoW2::WeightingType, DBoW2::ScoringType) [with TDescriptor = cv::Mat; F = DBoW2::FORB]’
/opt/ros/indigo/share/ORB_SLAM/src/main.cc:86:29:   required from here
/opt/ros/indigo/share/ORB_SLAM/Thirdparty/DBoW2/DBoW2/TemplatedVocabulary.h:446:3: warning: deleting object of abstract class type ‘DBoW2::GeneralScoring’ which has non-virtual destructor will cause undefined behaviour [-Wdelete-non-virtual-dtor]
   delete m_scoring_object;
   ^
[ 11%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/Tracking.cc.o
[ 16%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/LocalMapping.cc.o
[ 22%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/LoopClosing.cc.o
[ 27%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/ORBextractor.cc.o
/opt/ros/indigo/share/ORB_SLAM/src/ORBextractor.cc: In member function ‘void ORB_SLAM::ORBextractor::ComputeKeyPoints(std::vector<std::vector<cv::KeyPoint> >&)’:
/opt/ros/indigo/share/ORB_SLAM/src/ORBextractor.cc:606:63: error: ‘FAST’ was not declared in this scope
                 FAST(cellImage,cellKeyPoints[i][j],fastTh,true);
                                                               ^
/opt/ros/indigo/share/ORB_SLAM/src/ORBextractor.cc:615:34: error: ‘ORB’ has not been declared
                 if( scoreType == ORB::HARRIS_SCORE )
                                  ^
/opt/ros/indigo/share/ORB_SLAM/src/ORBextractor.cc:682:17: error: ‘KeyPointsFilter’ has not been declared
                 KeyPointsFilter::retainBest(keysCell,nToRetain[i][j]);
                 ^
/opt/ros/indigo/share/ORB_SLAM/src/ORBextractor.cc:698:13: error: ‘KeyPointsFilter’ has not been declared
             KeyPointsFilter::retainBest(keypoints,nDesiredFeatures);
             ^
/opt/ros/indigo/share/ORB_SLAM/src/ORBextractor.cc: In member function ‘void ORB_SLAM::ORBextractor::operator()(cv::InputArray, cv::InputArray, std::vector<cv::KeyPoint>&, cv::OutputArray)’:
/opt/ros/indigo/share/ORB_SLAM/src/ORBextractor.cc:759:82: error: ‘GaussianBlur’ was not declared in this scope
         GaussianBlur(workingMat, workingMat, Size(7, 7), 2, 2, BORDER_REFLECT_101);
                                                                                  ^
/opt/ros/indigo/share/ORB_SLAM/src/ORBextractor.cc: In member function ‘void ORB_SLAM::ORBextractor::ComputePyramid(cv::Mat, cv::Mat)’:
/opt/ros/indigo/share/ORB_SLAM/src/ORBextractor.cc:799:78: error: ‘INTER_LINEAR’ was not declared in this scope
             resize(mvImagePyramid[level-1], mvImagePyramid[level], sz, 0, 0, INTER_LINEAR);
                                                                              ^
/opt/ros/indigo/share/ORB_SLAM/src/ORBextractor.cc:799:90: error: ‘resize’ was not declared in this scope
             resize(mvImagePyramid[level-1], mvImagePyramid[level], sz, 0, 0, INTER_LINEAR);
                                                                                          ^
/opt/ros/indigo/share/ORB_SLAM/src/ORBextractor.cc:802:80: error: ‘INTER_NEAREST’ was not declared in this scope
                 resize(mvMaskPyramid[level-1], mvMaskPyramid[level], sz, 0, 0, INTER_NEAREST);
                                                                                ^
make[2]: *** [CMakeFiles/ORB_SLAM.dir/src/ORBextractor.cc.o] Error 1
make[1]: *** [CMakeFiles/ORB_SLAM.dir/all] Error 2
make: *** [all] Error 2

jwcrawley, 2015-07-07:
I personally solved it by loading synaptic, searching for ros-indigo and selecting every package that was installed. Then I marked them for complete removal, and then apply.

Then go to the bottom of your ~/.bashrc and remove all lines related to ROS.

And now, go through the install procedure as listed on ROS website.

That solved the problem of OpenCV libs not working for me.

aaronma37, 2015-08-13:
I was able to fix this by adding 
#include <opencv2/opencv.hpp>

to the top of ORB_SLAM/src/ORBextractor.cc.   My opencv library didn't have the header files

#include <opencv2/core/core.hpp>
#include <opencv2/highgui/highgui.hpp>

which were in ORB_SLAM/src/ORBextractor.cc, but after looking I found that #include <opencv2/opencv.hpp> had them.  I think this was a problem for me because I am trying to run 15.04 on jade

marcelinomalmeidan, 2016-02-02:
I had the same problem in 14.04 on Indigo. aaronma37's suggestion solved my problem!

yanyan-li, 2017-06-19:
aaronma37's suggestion solved same problem in 16.04

JeffYoung17, 2017-07-14:
I had same problem with aaronma37,and his method works.
My ubuntu is 14.04.

gandhiRobo, 2018-04-22:
aaronma37 solution works for 16.04Ubuntu on Kinetic

ckchng, 2019-11-04:
aaronma37 solution works for 18.04Ubuntu on Melodic