raulmur/ORB_SLAM #45: Error Make Indigo

NilsL3, 2015-07-07:
Hello,

I am currently using Ubuntu 14.04 + ROS Indigo. I  have cloned the ORB_SLAM repository to my catkin workspace and all of the necessary dependencies are installed, but when I launch "ExampleGroovyHydro.launch" then I get the following error:

ERROR: cannot launch node of type [ORB_SLAM/ORB_SLAM]: can't locate node [ORB_SLAM] in package [ORB_SLAM]

And indeed I cannot find the ORB_SLAM node within the package. Also when I run the following instruction: rosrun ORB_SLAM ORB_SLAM PATH_TO_VOCABULARY PATH_TO_SETTINGS_FILE, I get:

[rosrun] Couldn't find executable named ORB_SLAM below /home/nils/catkin_ws/src/ORB_SLAM.

Does anyone know how I could solve this error? I would appreciate it a lot. Regards.

weblucas, 2015-07-09:
Test another fork with a indigo-devel branch. I am using srv( https://github.com/srv/ORB_SLAM.git
) repository and it is working well.

NilsL3, 2015-07-14:
Thanks a lot for the help. I am currently trying this fork and I have bumped into a problem when launching the orb_slam.launch file. I obtain the following in the terminal:

Vocabulary loaded!

Camera Parameters: 
- fps: 30
- color order: RGB (ignored if grayscale)
- Images are rectified: No
- camera: MONO

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Fast Threshold: 20
- Score: FAST

However, the window titled "/image", which appears when running the launch file is grey and the RViz opens and mentions the following warning:

Global Status: Warn
Fixed Frame
No tf data. Actual error: Fixed Frame [ORB_SLAM/Camera] does not exist

I suppose I can't get it work because I didn't edit the launch file correctly. I was not sure where to insert the topic I was receiving the raw video stream from. Could you please tell me where in the launch file I should insert the topic I am subscribing to in order to receive the raw video stream? I would appreciate it a lot if you could tell me what I am doing wrong based on the error information above or if I should launch orb_slam differently.

weblucas, 2015-07-15:
I suggest you start first with the example sequence from the readme. With
this sequence you can use the launch file bellow. and play the "rosbag play
Example.bag"

=============================

<launch>

<node pkg="image_view" type="image_view" name="image_view" respawn="false"
output="log">
<remap from="/image" to="/ORB_SLAM/Frame" />
<param name="autosize" value="true"/>
</node>

<node pkg="rviz" type="rviz" name="rviz" args="-d $(find orb_slam)/Data/rviz
.rviz" output="log"/>

  <node pkg="orb_slam" type="orb_slam" name="orb_slam"  args="Data/ORBvoc.
yml Data/Settings.yaml" cwd="node" output="screen">

</node>

</launch>

========================

When you manage to run with this. Next step is to try your own data

you basically need two things. Calibration parameters in the Opencv
fashion. (use the ros calibrator - http://wiki.ros.org/camera_calibration).
These parameters go in the file (Data/Settings.yaml)
the second thing is remap to the input of your camera.
you need to uncomment the line  and change the parameter "to" with your input address. The
Example.bag publish its image already with the name expected by the
ORB-SLAM. this is why to run the example the remap is commented.

Good Luck

NilsL3, 2015-07-15:
Thanks a lot for the detailed explanation. I'll keep you up to date on how I'm doing.

NilsL3, 2015-07-15:
I have tried executing orb_slam with the provided dataset, but I still have had no success. I followed your instructions, I made the according modifications to the launch file and then I used "roslaunch" to execute the launch file. And then I used "rosbag play" to play the dataset, but the image view window now titled "ORB_SLAM/Frame" is black and in the bottom corner a text mentions the following: "WAITING FOR IMAGES. (Topic: /camera/image_raw)". 

The Rviz window is blank and I still have the following error: 
Global Status: Warn
Fixed Frame
No tf data. Actual error: Fixed Frame [ORB_SLAM/Camera] does not exist

When playing the dataset sequence, there are no changes neither in the image view window nor Rviz.

weblucas, 2015-07-15:
Use the rqt application to double check that your bag is playing and see
the name of the topic

NilsL3, 2015-07-16:
I have used the rqt application and I can view the dataset sequence on the "/camera/image_raw" topic, however the "ORB_SLAM/Frame" image view window remains black and in the bottom corner of this window a message announces the following "WAITING FOR IMAGES. (Topic: /camera/image_raw)".

NilsL3, 2015-07-16:
I tried a different fork and I got orb_slam to work with the dataset sequence. Thanks a lot for the help.

weblucas, 2015-07-16:
No problem.

mihaibujanca, 2015-08-21:
I am having similar problems. Camera is working on camera/image_raw and I can view ORB_SLAM/Frame using image_view, but I am still getting `No tf data. Actual error: Fixed Frame [ORB_SLAM/Camera] does not exist` with rviz.

pvmilk, 2015-09-12:
Just incase anyone encounter the same issue. 

Make sure the lunch file (ExampleGroovyOrNewer.launch) load the type 'orb_slam' not 'ORB_SLAM'

<node pkg="ORB_SLAM" type="orb_slam" name="orb_slam"  args="Data/ORBvoc.txt Data/Settings.yaml" cwd="node" output="screen">

There will be a few more errors regarding the libraries of the Thirdparty.
Simply add the path to your LD_LIBRARY_PATH.

pmoulon, 2015-11-01:
@weblucas, 
I cannot connect my webcam. I have also the message "No tf data. Actual error: Fixed Frame [ORB_SLAM/Camera] does not exist" in rviz.
@mihaibujanca  -> Do you have found the solution to your problem?

here my Launcher.

```
<launch>

  <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video0" />
    <param name="image_width" value="640" />
    <param name="image_height" value="480" />
    <param name="pixel_format" value="mjpeg" />
    <param name="camera_frame_id" value="usb_cam" />
    <param name="io_method" value="mmap"/>
    <param name="framerate" value="30"/>
  </node>
  
  <node pkg="image_view" type="image_view" name="image_view" respawn="false" output="log">
	<remap from="/image" to="/ORB_SLAM/Frame" />
	<param name="autosize" value="true"/>
	</node>

	<node pkg="rviz" type="rviz" name="rviz" args="-d $(find ORB_SLAM)/Data/rviz.rviz" output="log">
   </node>

 	<node pkg="ORB_SLAM" type="ORB_SLAM" name="ORB_SLAM"  args="Data/ORBvoc.txt Data/Settings.yaml" cwd="node" output="screen">
 	<remap from="/camera/image_raw" to="/usb_cam/image_raw" />
   </node>
   
</launch>
```

weblucas, 2015-11-02:
Hi Pierre,

To fix this error you have to change in the Global Options-> Fixed Frame to
another option on the combo box while you are running the ORB-SLAM.

Best

Lucas

mhzaki, 2016-01-17:
Hello,
hey @NilsL3 can you give me the fork which it works for you.? because i still have the same problem  than you .. 
thank you.

NilsL3, 2016-01-17:
Hello mhzaki,

The fork which worked for me is the following: bmagyar/ORB_SLAM
I hope it helps.

mhzaki, 2016-01-20:
hello @NilsL3 
I this fork but i still have the same problem i don't know, I am using ros hydro on ubuntu 12.04. 
Did you cactinized it or use the way described in the readme file ?

NilsL3, 2016-01-20:
Hi @mhzaki 
I used this fork with Ubuntu 14.04, ROS Indigo, and I used it just as it is described in the readme file.

KathiravanNatarajan, 2017-02-04:
@NilsL3 how is it exactly working ? I have done the setups mentioned in the README.md file. 

can you explain briefly about your observation ??

KathiravanNatarajan, 2017-02-04:
@weblucas : Hi..I have done with the setups mentioned. what is the vocab file ? I couldn't understand certain things..you seemed to have good knowledge on it. Could you explain the complete process ? I am ready with the setups