raulmur/ORB_SLAM #46: Location of probabilistic semi-dense mapping module

jwcrawley, 2015-07-08:
I am trying to replicate and use your work as displayed in said paper and front youtube video.

https://www.youtube.com/watch?v=HlBmq70LKrQ

Have you released this module? If so, how do I call it appropriately? If not, what is your timeframe in which you will release it in this github?

Sincerely,
Josh Conway

bjornph, 2015-07-10:
I asked about it some weeks ago. This was the answer:

"We may release the code in the future, but I will not be soon.
Thanks for your email.

Best regards,
Raul"

LSD-SLAM seems to be the only available semi-dense mapping algorithm as of now: http://vision.in.tum.de/research/lsdslam

jwcrawley, 2015-07-13:
That's very unfortunate, considering this is a feature mentioned on the ORB-SLAM front page.

I believe that I am not the only one who cares about sem-dense mapping, so I'll keep this open as a bug to warn others that ORB-SLAM does not have this feature.

NikolausDemmel, 2015-07-13:
It's not like Raul has some obligation to make his code public...

jwcrawley, 2015-07-13:
@NikolausDemmel  Absolutely true. But I wouldn't have spent any time on this had I known that semi-dense was not included. LSD-SLAM and rolling your own seems to be the two main ways at this time.

I'm going to keep this opened so it alerts people that expected functionality isn't included.

yash508oza, 2016-10-25:
Have they released it yet?

jwcrawley, 2016-10-25:
Far as I've looked at it, they still haven't.

I've moved on to LSD slam. You get what you see with that project. There's
no obfuscation or misdirection of what is really offered.

Josh

raulmur, 2016-10-26:
@yash508oza we do not have plans to release the code for now. If you really need semi-dense reconstructions then you can try [LSD-SLAM](https://github.com/tum-vision/lsd_slam) or [DPPTAM](https://github.com/alejocb/dpptam).

closed 2016-10-26 by raulmur