raulmur/ORB_SLAM #48: used equipement

5aou, 2015-07-22:
Hi,
Can you tell me please witch camera (and other equipments,PC.... ) did you use, thank you.
Best regards.

weblucas, 2015-07-22:
my favorite is the mvBlueFOX-MLC200w with BM2420C
pc - lenovo y510p

5aou, 2015-07-22:
ok thank you a lot.

5aou, 2015-07-22:
Can other users tell me witch equipments they use plz !

weblucas, 2015-07-22:
Did you not like my camera?

5aou, 2015-07-22:
that's not the point, I have to buy materials for my project , I'm using a logitech cam and it doesn't work,that's why I want have more informations about all equipments that work well and without errors with orb-slam.

bjornph, 2015-07-22:
^ Which Logitech camera are you using? I have gotten decent indoor results with Logitech c310 and c920. It has been answered before that it's recommended to use a global shutter camera with a wide angle lens, see https://github.com/raulmur/ORB_SLAM/issues/24.

However it is possible to get decent results with a consumer grade web camera as well. Be sure to calibrate it correctly (http://wiki.ros.org/camera_calibration). It is important to calibrate in a well lit room as low light rooms might cause the blurry images. The TUM RGB-D datasets are all recorded on Kinect and Asus Xtion cameras, where the RGB part is nothing better than most affordable web cameras.

While it might be worth it to invest in a more expensive camera unit, I would advise you to make sure that your system is running correctly first.

5aou, 2015-07-23:
thank you for you response,I'm using a  Logitech QuickCam S5500, I managed to calibrate it by installing "ros-indigo-usb-cam"' , but I cant resolve the problem mentioned at #42  , do you have some suggestions ?

bjornph, 2015-07-24:
OK. I don't think that's a good camera, but you should be able to get some results that at least makes some kind of sense. Also see my answer which I will give you in #42 now

jingpang, 2015-12-12:
@5aou  Maybe it's a little late now about your question as you may have purchased one, but i think ueye camera is a good choice.
(https://en.ids-imaging.com/store/produkte/kameras/usb-2-0-kameras/ueye-le.html)

amiltonwong, 2015-12-14:
@jingpang ,
I also have the similar model, which is iDS UI-1221LE
http://en.ids-imaging.com/store/ui-1221le.html

But I came across the following issue.
When it is used in outdoor scenario, it suffers from overexposure issue, in which I only see the plain white screen under daylight. But it is OK when it is used in indoor scenario under moderate lights.

Did you come across when you use iDS monocular camera? And how do you handle it?

THX~
Milton

jingpang, 2015-12-16:
@amiltonwong ,
You can try auto exposure instead of fixed.

I use the ueye_cam package to read frames.
(sudo apt-get install ros-indigo-ueye-cam , see 'http://wiki.ros.org/ueye_cam'  ) 

Below is my ros launch file in which the exposure and other modes can be configured.
(https://github.com/jingpang/ORBSLAM_jpMiniPC/blob/master/launch/ueye_rgb8_settings.launch)

amiltonwong, 2015-12-16:
Thx @jingpang , I'll try it.