raulmur/ORB_SLAM #49: ORB-SLAM dies after 5 minutes

MarcGyongyosi, 2015-08-04:
Hi,

when using ORB-SLAM in our applications (onboard an MAV) everything works fine until about 5 minutes into flight. Usually around that time, the ORB-SLAM process dies. Is there a way to see/check why it dies? Is it a memory issue? It just gives us exit code -9

raulmur, 2015-08-04:
Hi, 
Maybe the system runs out of memory and the OS kills the process. Can you check the memory consumption when you are running the application? How much RAM do you have in the MAV?

MarcGyongyosi, 2015-08-04:
Hi thanks for getting back. I will check the memory consumption during runtime when i get back to the lab later today. I certainly know that I have 2Gb RAM onboard.

raulmur, 2015-08-04:
Ok, 2Gb is extremely little RAM for ORB-SLAM. The current way to load the ORB vocabulary through cv::FileStorage requieres a bit less than 2Gb (this could be done much more efficiently, but it would requiere you to modify the save/load vocabulary functions of DBoW2). Another option is to use a smaller vocabulary ( 5 levels and10 nodes per level, should work fine as well, currently we are using 6 levels).

MarcGyongyosi, 2015-08-04:
Ok thanks, I created a smaller vocabulary and it seems to do better now - it's been running for about 7 minutes so far, no problem yet! I was wondering if you could elaborate on what you consider a "large" set of images (as you write in the paper). What number of images do you recommend for a fixed indoor environment? And do you recommend those pictures only be taken in the respective indoor environment or does including pictures from other places improve tracking accuracy?

Thanks!

Marc

raulmur, 2015-09-01:
Check the new update of the code. We load the vocabulary in a more efficient way and now it requires only about 350Mb.

MarcGyongyosi, 2015-09-11:
I will check this out next week, just got back from vacation. Thanks for publishing the modification!

MarcGyongyosi, 2015-09-29:
yes, it works much better now. Thanks for making this fix available!

closed 2015-09-29 by MarcGyongyosi