raulmur/ORB_SLAM #50: error: Wrong path to settings

385693162, 2015-08-06:
When I launch ORB-SLAM from the terminal by "rosrun ORB_SLAM ORB_SLAM PATH_TO_VOCABULARY PATH_TO_SETTINGS_FILE",There will be a mistake - [ERROR] [1438841720.919284826]: Wrong path to settings. Path must be absolut or relative to ORB_SLAM package directory .
ps：I have  provided the path of  ORB_SLAM directory to the environment variable of  PATH_TO_SETTINGS_FILE

gitunit, 2015-08-06:
uncompress zip file and it should be solved. the documentation (section 4, point 1) says: 

"You have to provide the path to the ORB vocabulary and to the settings file. The paths must be absolute or relative to the ORB_SLAM directory.
We already provide the vocabulary file we use in ORB_SLAM/Data/ORBvoc.yml. Uncompress the file, as it will be loaded much faster."

385693162, 2015-08-06:
Thank you very much for replying me.I have uncompress the ORBvoc.yml.I guess is whether their(PATH_TO_VOCABULARY PATH_TO_SETTINGS_FILE) path to write wrong.I would like to ask you whether Their path(PATH_TO_VOCABULARY PATH_TO_SETTINGS_FILE) is the same ? and can only write one path to  environment variable ? 
[image]
[image]

Po-Jen, 2015-08-09:
Here's my scripts and my launch file:

https://github.com/Po-Jen/scripts/blob/master/orb_slam/slam.sh
https://gist.github.com/Po-Jen/1dc24fde19727407c3a8

You can see how to specify the path in the last of the launch file.

385693162, 2015-08-09:
thank you very much!!!

385693162, 2015-08-09:
用他提供的rosbag可以运行了。可是我想用自己的内置摄像头，当执行  rosrun ORB_SLAM ORB_SLAM PATH_TO_VOCABULARY PATH_TO_SETTINGS_FILE  会出现如下错误：
[image]

Po-Jen, 2015-08-09:
PATH_TO_VOCABULARY and PATH_TO_SETTINGS_FILE means nothing to your program. You should give it the real path. You should know that since you have successfully run the rosbag example.

385693162, 2015-08-10:
原來如此啊！
那麼你給的這兩行代碼是打開攝像機然後將數據放入/camera/image_raw這個topic中去是嗎 ？
（roslaunch /home/rosfuerte/project/ros/pc_cam.launch /usb_cam/image_raw:=/camera/image_raw
roslaunch ORB_SLAM ExampleFuerte.launch）

那麼我的攝像機節點是下面這個吧
[image]

我運行下面的命令：roslaunch /home/chenhan/catkin_ws/src/usb_cam/launch/usb_cam-test.launch /usb_cam/image_raw:=/camera/image_raw
[image]
這個錯誤，我不太理解
/usb_cam/image_raw:=/camera/image_raw  這一句

385693162, 2015-08-11:
How can I solve this problem of "No tf data. Actual error: Fixed Frame [ORB_SLAM/Camera] does not exi"  
[image]

Po-Jen, 2015-08-11:
You need to use your own camera driver, usb_cam is the driver I am using. It complains that the executable usb_cam_node does not exist in your computer.

If you have your camera publish image successfully, you can start moving around. Once you do enough frames, it'll then switch from lines to little squares showing keypoint mappings. The error of "No tf data. Actual error: Fixed Frame [ORB_SLAM/Camera] does not exists" would then disappear.

You can check jwcrawley's comment in [issue-42](https://github.com/raulmur/ORB_SLAM/issues/42).

385693162, 2015-08-12:
感谢啊！虽然摄像机没有显示图像，但是可以实现map了。
我现在先利用这些点云进行三维重建，请问您觉得这样可行吗？那些点云是可以提取出来的吧？

weblucas, 2015-08-12:
啊！虽摄像没显示图

Po-Jen, 2015-08-12:
If you have seen feature points shown in Rviz, I think you can progress.

You should note that the map you created are sparse, if you want a semi-dense map, you should take a look at LSD-SLAM.

385693162, 2015-08-13:
十分感谢！！！开始看LSD的paper。

Po-Jen, 2015-08-14:
If the problem is solved, you should close this issue. Thx~

closed 2015-08-14 by 385693162

ldhl103, 2015-09-01:
the newer version is the ORB_SLAM/Data/ORBvoc.txt.tar.gz.
I uncompressed it, and my command is :

rosrun ORB_SLAM ORB_SLAM Data/ORBvoc.txt Data/settings.yaml

but I met the same problem 

 Wrong path to settings. Path must be absolut or relative to ORB_SLAM package directory.

how do I solve it?

raulmur, 2015-09-01:
The default settings file is "Data/Settings.yaml" not "Data/settings.yaml"
rosrun ORB_SLAM ORB_SLAM Data/ORBvoc.txt Data/Settings.yaml

In any case, check the system is working by running the example sequence using the Launch file provided (see section 5 in the documentation).

LEETZUNGYOU, 2016-01-15:
Hello @Po-Jen  May I ask a question?
Why when I run Example.bag My ORB-SLAM didn't work?

Po-Jen, 2016-01-15:
You need to provide the steps you did and how did you fail, thx.

marcelinomalmeidan, 2016-02-03:
1st: I really wish I could understand whatever was written here before the "If the problem is solved, you should close this issue. Thx~". Spoken language didn't help =/

2nd: I am having similar problems as described here. I followed the instructions in Readme.md, and seemed to succeed up to section 3. If I did understand Section 4 well, I have to execute the line:
rosrun ORB_SLAM ORB_SLAM PATH_TO_VOCABULARY PATH_TO_SETTINGS_FILE

I have an open terminal with roscore executing, and I open a new terminal to execute the above, which in my case is:
rosrun ORB_SLAM ORB_SLAM /Data/ORBvoc.txt /Data/Settings.yaml

The output is:
ORB-SLAM Copyright (C) 2014 Raul Mur-Artal
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.
Segmentation fault (core dumped)

I don't understand what this Segmentation fault is. And yes, I did extract the .tar.gz file.

Still a little bit confused, I tried to go to section 5 (this forum thread made me believe that I had go through Section 5 straight away). I downloaded the example file, went to the ORB_SLAM folder in the terminal and typed (I use 14.04 with Indigo):
roslaunch ExampleGroovyOrNewer.launch

The output is long, but it seems that the error is the following:
[ERROR] [1454457475.590683661]: Wrong path to settings. Path must be absolut or relative to ORB_SLAM package directory.
[ORB_SLAM-3] process has died [pid 16098, exit code -11, cmd /home/user/ORB_SLAM/bin/ORB_SLAM Data/ORBvoc.txt Data/Settings.yaml __name:=ORB_SLAM __log:=/home/user/.ros/log/f70f21a6-c9c6-11e5-8896-0023aea65975/ORB_SLAM-3.log].
log file: /home/user/.ros/log/f70f21a6-c9c6-11e5-8896-0023aea65975/ORB_SLAM-3*.log

I apologize if this was supposed to be easy to solve, but I wasn't able to understand well what to do from the Readme file.

MarcGyongyosi, 2016-02-03:
either remove the "/" from before the paths and make sure you're executing in the right directory, or give it absolute paths. if you give it /Data it will try to find the files in the directory /Data on your computer and not within the ORB SLAM directory

marcelinomalmeidan, 2016-02-03:
Thank you MarcGyongyosi! It makes sense what you said, but I just ran (I am presently in the ORB_SLAM folder):
rosrun ORB_SLAM ORB_SLAM Data/ORBvoc.txt Data/Settings.yaml

Output (Same as before =/):
ORB-SLAM Copyright (C) 2014 Raul Mur-Artal
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.
Segmentation fault (core dumped)

ram651991, 2016-05-26:
I'm not providing a solution here, but I just would like to share a similar scenario i faced while running rgbdslam_v2 package. So I would like to say that this may or may not be the exact solution to your problem. "Segmentation fault (core dumped)" occurs generally due to memory issues, as in, either you're out of memory or the package is trying to access a memory area it is not supposed to. Lately i found out that I hadn't declared a swap area in my ubuntu machine where the rbgdslam package wasn't having enough memory to run itself and hence threw the "Segmentation fault" error. Off late I got the same error when i tried to run the package in raspberry pi a+ (which has only 256MB RAM). Once I fixed the swap area (by providing enough swap memory - 8gb is what i allocated), I didn't get the Segmentation fault error either in my Ubuntu machine or raspberry pi.

chaiein, 2016-11-29:
[image]

There is enough space but still segmentation fault

gs14iitbbs, 2017-03-22:
Helllo @marcelinomalmeidan, how did you solve your problem ?
I am having the exactly same situation. 
I tried both  rosrun ORB_SLAM ORB_SLAM Data/ORBvoc.txt Data/Settings.yaml
and  roslaunch ExampleGroovyOrNewer.launch
but none is working as shown below:

[image]

[image]

Please let me know how did you solve it.
Is it problem with opencv version compatibility or something else?

marcelinomalmeidan, 2017-03-22:
Hey @gs14iitbbs! Sorry, its been so long, I don't quite recall how I solved it. I feel bad that I didn't post my solution here =/
However, I recall uninstalling OpenCV and reinstalling the right version, which helped for something. I just don't recall if this actually helped for this one problem.

gs14iitbbs, 2017-03-23:
Okay !! thanks, I will try to solve it changing OpenCV version. If I could solve it, I will post it.

latifAnjum, 2017-08-23:
My launch command:
`rosrun ORB_SLAM ORB_SLAM Data/ORBvoc.yml Data/Settings.yaml`
Generates this error.
[image]

System Info:
I have Ubuntu 16.04 with ROS Kinetic Kame. Please help. I have successfully built ORBSLAM in ROS Kinetic Kame, so I guess this distro of ROS is not the problem.

wangzhaoguang, 2017-09-13:
@latifAnjum 
your command is not right : rosrun ORB_SLAM ORB_SLAM Data/ORBvoc.yml Data/Settings.yaml
 should like : rosrun ORB_SLAM ORB_SLAM Data/ORBvoc.txt Data/Settings.yaml
anyway, even run the above one, I met errors, saying "Wrong path to settings. Path must be absolut or relative to ORB_SLAM package directory."

wangzhaoguang, 2017-09-13:
@ChenhanXmu hello chen, you got through the error but you did not say how.
actually I met the same problem, like below: 
[image]