raulmur/ORB_SLAM #51: multiple camera with orb_slam

navidmostofi, 2015-08-11:
Hi

I have a question about how to integrate multiple cameras in ORB_SLAM in way that each can use the map generated from the other one (not generate new map) for localization. 

Thanks

aaronma37, 2015-08-13:
Hi Navid,

I'm not sure if anyone has made anything yet, but I was able to get multiple cameras tracked and publishing new keyframes.  If you modify the tracking.cc file to accept multiple camera streams and make multiple instances of "trackers" it works pretty well.  Let me know if I can share it

Oblynx, 2015-08-13:
Hi aaronma37,

Combining many cameras is a very interesting project. I think tracking is
the first step and then you can try to combine the data they gather for the
mapping.
Do you work on a fork of ORB_SLAM?

*-------------------------------------------------------*
*  Konstantinos Samaras-Tsakiris*
School of Electrical and Computer Engineering
Aristotle University of Thessaloniki

aaronma37, 2015-08-13:
Hey Oblynx,  I will upload it to a fork later today.  I am a little new to github so I didn't want to mess anything up.

navidmostofi, 2015-09-09:
Hi,
anyone did stereo matching with ORB_SLAM? I want to remove the current initialization part by depth entimation from stereo but the problem is the matching of ORB betwen left and right image fail, anyone has any similar experience?

waynepa, 2016-12-14:
Hi @aaronma37  , 
I tried to run your code. But i can't work for 2 camera.
Do you have more information about this project?