raulmur/ORB_SLAM #53: Using ATAN camera model

ssampang, 2015-08-16:
I'm using a very wide FOV camera (go pro), so I'd like to use the ATAN camera model instead of the provided model that OpenCV uses. I believe I've modified the code correctly, as I've double-checked my math by hand (by observing and verifying the final undistorted coordinates I feed back into the ORB-SLAM code) and the features have no problems tracking near the center of the frame. However, I'm having trouble with disappearing features at the edges of the frame (maybe the first 20% of each edge), so I wanted to check here and see if the authors had maybe implemented some checks to delete features that were corrected by the calibration to be too far outside of the image bounds eg. a feature at (10,10) in an image of size (1920,1080) that was then corrected to (-150,-50).

Thanks,
Sid

Oblynx, 2015-08-16:
Hello Sid,

Do you have your modifications in a fork here on Github by any chance? I'd
love to take a look.

Thanks,
Konstantinos

*-------------------------------------------------------*
*  Konstantinos Samaras-Tsakiris*
School of Electrical and Computer Engineering
Aristotle University of Thessaloniki

HemangShah1, 2015-08-16:
Hi,

Each map point gets checked whether it is in frustum (angle is checked). See the function: Frame::isInFrustum.

hth,

ssampang, 2015-08-17:
Hi guys,

@Oblynx Oblynx: Unfortunately, I can't share the code without approval from my employer, which is a long process.

@HemangShah1: Thanks for the tip! Looking at `Frame::isInFrustum` led me to look at `Frame::ComputeImageBounds`, which was also calling `cv::undistortPoints`. I simply had to call my function there as well, and the code works now. 

Thanks again,
Sid

closed 2015-08-17 by ssampang