raulmur/ORB_SLAM #55: Code for learning vocab tree

nniksan, 2015-08-22:
@raulmur: 

Hi,
I was wondering if you could point me to the code you used to train the vocab tree. Is it something you put together or is it part of Dbow2?
Thanks
Nikhil

raulmur, 2015-08-24:
Hi,
You can create the vocabulary with the "create" function in the TemplatedVocabulary.h from DBoW2:
https://github.com/dorian3d/DBoW2/blob/master/include/DBoW2/TemplatedVocabulary.h

Best,
Raúl

closed 2015-08-24 by raulmur

enddl22, 2015-10-27:
Hi, Raul
In order to use the create function, feature vector (vector<vector<TDescriptor> >) is required as an input. I have looked at https://github.com/dorian3d/DBoW2 and loadFeatures function from demo.cpp provides some hints for generating feature vector. But it is for surf feature. 
For ORB feature, is the following attempt feasible?
```cpp
void loadFeatures(vector<vector<vector<float> > > &features)
{
         cv::Ptr<cv::ORB> orb = new cv::ORB(1000);

         //Load an image

         (*orb)(image, mask, keypoints, descriptors);

         features.push_back(vector<vector<float> >());

         changeStructure(descriptors, features.back(), orb->descriptorSize());
}
```
and some parameters
```cpp
  const int k = 10;
  const int L = 6;
  const WeightingType weight = TF_IDF;
  const ScoringType score = L1_NORM;
```
It is also great if you can share "loadFeatures" function that you used for generating ORB feature vectors.
Cheers,