raulmur/ORB_SLAM #58: OpenCV ERROR: Parsing error after rosrun  ORB_SLAM

ldhl103, 2015-09-01:
Hi I am a new programmer ... i want to try to solve this

after i do

rosrun ORB_SLAM ORB_SLAM PATH_TO_VOCABULARY PATH_TO_SETTINGS_FILE

I got this:

Loading ORB Vocabulary. This could take a while.
OpenCV Error: Parsing error (/home/huang/thesis/ORB_SLAM/Data/ORBvoc.txt(1): Valid XML should start with '<?xml ...?>') in icvXMLParse, file /tmp/buildd/ros-fuerte-opencv2-2.4.2-1precise-20130312-1306/modules/core/src/persistence.cpp, line 2257
terminate called after throwing an instance of 'cv::Exception'
  what():  /tmp/buildd/ros-fuerte-opencv2-2.4.2-1precise-20130312-1306/modules/core/src/persistence.cpp:2257: error: (-212) /home/huang/thesis/ORB_SLAM/Data/ORBvoc.txt(1): Valid XML should start with '<?xml ...?>' in function icvXMLParse

I need someone help me

raulmur, 2015-09-01:
It seems you are using an old version of main.cc, where we loaded the vocabulary file using cv::FileStorage. Please make sure you are using the last version of the code.

ldhl103, 2015-09-01:
Hi @raulmur 

I just did git clone this repository two days ago, and I tried it again.
the problem still exists. Isn't it the last version?

Or does it depend on my OpenCV version? I'm used to using OpenCV 2.4.9

raulmur, 2015-09-01:
I have updated the code and the vocabulary today. The error says you are loading ORBvoc.txt , which is the new vocabulary (before it was ORBvoc.yml). So you are mixing the new vocabulary with the old code.