raulmur/ORB_SLAM #59: Tracking lost shortly after initialization

sneuma12, 2015-09-03:
Hi Raul,
I'm working with ORB-SLAM on a powerful Windows-PC without ROS. After a while dealing with compilation errors it is running now (g2o made some problems). I have entered the calibration data from my camera and it works. The problem is, that the system loses its tracking pose very often, although my video sequence has a very smooth translation and there's a lot of thankful features in the frames.

The reason for the loss is that the "mnMatchesInliers" falls fast below the threshold.
The reason for the small number of Matches lies in the fact that the number of Map-Points do not really grow after the initialization.

Do you have an idea why that could happen ? If you need more information about my situation feel free to ask :)

Best Regards,
Sergej

raulmur, 2015-09-04:
Have you tried the example sequence?. Does it work correctly?

You can send me some seconds of your sequence and the calibration and I can take a look.

sneuma12, 2015-09-04:
No, i haven't tried the example sequence. I will try it today and give some feedback about the results. Thanks for the hint.

sneuma12, 2015-09-04:
I found my mistake. Something was wrong with the vocabulary. After updating the new "Vocabulary read in method" it works very good and robust.

the only thing i have to admit is that it runs quite slow. It processes about 6 images per seconds. Is it possible that its slower because it runs under Windows and the Visual C++ 2005 Compiler ?

Beewe, 2015-09-05:
I'm running my build on Win7 (but with VS2013) and can easily process more than 30fps on my PC and around 25 on my Laptop (also VS2013), so it may depend on the 2005 Compiler. Maybe you can make sure you build all libraries and your application in Release Mode, so with all optimizations turned on?

Sunny-cpp, 2018-04-13:
hi, Sergej:
       I seem come across the same problem as you!
       but what is "updating the new Vocabulary read in method" mean?
        
        hope your reply
            best Regards