raulmur/ORB_SLAM #60: Problems while running with Automotive Sequence

DannyTheDog, 2015-09-07:
Hi Raul, 

thank you for the good job. 

## Basic Info 
I’m using Ros Indigo on a Ubuntu 14.04 

## Problem 
At the moment I have tested your code on the sequence inside the lab you provided and it worked well. 
Now I am trying to use it on an automotive sequences (I have generated the Rosbags using your tool <a href="https://github.com/raulmur/BagFromImages" target="_blank">BagFromImages</a>) and I have experimented an important performance degradation wrt the previous test: the results I have seen are significantly different from the KITTI test videos you posted. 

The sequences I am working on should not be part of the “Failure Modes” 

Could this problem be related to an incorrect `Settings.yaml` parametrization? 
Calibration params have been set correctly. 
Images that I’m using have been previously dedistorted hence k1,k2,p1,p2 have been set to zero. 
Camera.fps has been set according to the Rosbag fps creation param to 15 fps. 

Without any further tuning, running the sequence seems to result in a failing feature tracking hence no map is created. 

## Tested Solution 
I tried to fix the problem adjusting the following parameters: reducing ORBextractor.fastTh (less restrictive) and increasing ORBextractor.nFeatures (more features) and feature tracking starts working. 
The map gets created and localization is performed. 

## Persistent Problem 
After a couple of minutes of running, suddenly all the features disappear and never come back. No logging error message is written hence I have no clue about what the problem is. 
As far as I know there is no way to recover from such a situation and I have to restart the app. 

Thank you in advance for any help you could provide.

raulmur, 2015-09-08:
In the persistent problem, you say that features disappear and there is no logging error. Does ORB_SLAM/Frame say "TRACK LOST"? 
Before that happens, the trajectory seems to be correct? (that could tell you if the calibration is correct or not).
Play the rosbag slower using option -r in rosbag play (for example rosbag play ROSBAG_NAME -r 0.2) . If the system works, that means your computer is not powerful enough to run that sequence in real-time.

DannyTheDog, 2015-09-08:
Hi Raul, 

the suggestion to play the Rosbag led to an improvement: now I’m not experiencing the previously described “feature detector sudden blackouts” 

However I’m still experiencing the following problems 

## Feature Tracking Problem 

### Description 

I had to keep the nFeature value at 2000 because at 1000 the feature tracking seems to be still failing (actually tracking seems to be performed on some features but only for a very low number of frames, probably only one, hence I assume no feature is considered stable and used as a map element) then no map is ever created. 

### Questions 

Have you used different `Settings.yaml` files for indoor and automotive sequences ? 

## Visualizer crash 

### Description 

After some minutes of running, the RViz crashes with the following message 

terminate called after throwing an instance of 'Ogre::ItemIdentityException'
  what():  OGRE EXCEPTION(5:ItemIdentityException): chainIndex out of bounds in BillboardChain::addChainElement at /build/buildd/ogre-1.8-1.8.1+dfsg/OgreMain/src/OgreBillboardChain.cpp (line 243)

It seems to be an Ogre specific problem (afaik the Ros dependency used as visualization engine) and after some search I have found 

<a href="https://github.com/ras-sight/hratc2015_framework/issues/4" target="_blank">rviz crash: terminate called after throwing an instance of 'Ogre::ItemIdentityException'</a>

The kind of error looks similar, the different lines could be due to the different Ogre Version (I’m using Ros Indigo with Ogre 1.8 while the reported issue is related to Ogre 1.7.4) 

### Workaround 

I have tried to run the app deactivating all the RViz Options (Camera, Map Points, Keyframes, Graph) and I have not experienced any RViz crash. 
Considering I have not read any erroneous log message, I assumed the map computation has been succesfully completed so I activated the above mentioned flags but the RViz Map Area remained empty hence I had no way to check resulting map quality. 

Moreover unfortunately I have not read any "Loop Closure Message" (the sequence contained a loop closure) hence, in the context of this workaround, I can not distinguish whether 
- no map has been computed (hence unflagging RViz options causes no map computation, but I don't think so)
- map has been computed but no loop closure has been detected 

### Questions 

Have you experienced a similar problem while creating the videos you posted ? 

Which version of Ros and Ogre did you use (and tested) the system ? 

## Extra 

Btw if you prefer I can open different issues for these problems as they are (very likely to be) independent 

Thank you very much in advance