raulmur/ORB_SLAM #62: SegFault in DBoW2::L1Scoring loading the default vocabulary

jlblancoc, 2015-09-08:
Hola Raúl!

Just to let you know that, under Ubuntu 14.04 (x64), the default vocabulary (extracted from ORBvoc.txt.tar.gz) seems to make the program crash just at start-up. 

gdb says:

```
Loading ORB Vocabulary. This could take a while.

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff54cf910 in DBoW2::L1Scoring::mustNormalize(DBoW2::LNorm&) const () from /home/jlblanco/catkin_ws/src/ORB_SLAM/Thirdparty/DBoW2/lib/libDBoW2.so
(gdb) 
```

I could try to investigate it, but perhaps you are more familiar with the code and find the problem much faster?

raulmur, 2015-09-08:
Hola José Luis,

I have tested ORB-SLAM in Ubuntu 14.04 (x64) and I have never received that error. I have updated the function to load the vocabulary in Thirdparty/DBoW2/DBoW2/TemplatedVocabulary.h, adding some checking before creating the ScoreObject that seems to produce the error. Can you please update your code and run it again?

Thanks!

DannyTheDog, 2015-09-08:
Hi Mr. Blanco-Claraco, 

also I have tried to replicate the error but I have not seen it. 
Have you used any special flag when compiling DBoW2 ?

jlblancoc, 2015-09-08:
I have gcc 4.8.4, just in case you use another version.

Thanks for the update, but the error is still 100% reproducible every time it is launched. 
BT from GDB:

```
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff54cf910 in DBoW2::L1Scoring::mustNormalize(DBoW2::LNorm&) const () from /home/jlblanco/catkin_ws/src/ORB_SLAM/Thirdparty/DBoW2/lib/libDBoW2.so
(gdb) bt
#0  0x00007ffff54cf910 in DBoW2::L1Scoring::mustNormalize(DBoW2::LNorm&) const () from /home/jlblanco/catkin_ws/src/ORB_SLAM/Thirdparty/DBoW2/lib/libDBoW2.so
#1  0x00000000004365c3 in DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>::createScoringObject() ()
#2  0x0000000000439ed2 in DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>::loadFromTextFile(std::string const&) ()
#3  0x000000000042d898 in main ()
(gdb) 
```

Anyway, I can "fix" the crash by adding an arbitrary std::cout trace before the call to createScoringObject() in TemplatedVocabulary.h... if that can be called a "fix" :-( 
I tried to run it 10 times and it crashes all the times without the "cout", it passes with it. 

For me it is fine to close the issue if nobody else can reproduce it.

Theizu, 2015-09-13:
i do have the same issue. I compile with corei7 instead of march=native