raulmur/ORB_SLAM #64: optimization problem after loop closure

gitunit, 2015-09-18:
in general, i find ORB_SLAM very accurate. good job there. 
however after testing (several month) and working with a 10 min. bagfile i've recorded, i can't get a fully accurate map. 2/3 of the map seems to be very well while the other third is not good optimized after loop closure. before loop closure the map seems to be fine except the drifts of course (especially scale) but the lines on the street and the buildings are recognisible, this is not the case for the last part of the map after loop closure. please note, that the graph (in green) looks very similar to the location in google maps, the problem is keyframe positions and points are misplaced after loop closure for the last part of the map.

so, my question is, how can this be improved? is the bagfile problematic (too much rotation or backwards movement)?

## but see for yourself:

#### before loop closure (please note the parallel lines to the graph on the lower part):

[image]

#### after loop closure:

[image]

#### problematic parts (the part in the rectangle, mappoints are misplaced, street lines and buildings not recognisible anymore):

[image]

#### maybe the problem arises because of the bad movement in the corner before the last part?

[image]

gitunit, 2015-09-21:
i will try to record the same place with a different camera and report back later if the same happens or not.

raulmur, 2015-09-21:
Loop closing is designed to compensate the drift accumulated during exploration. Under good operation, this drift is expected to be somehow uniform along the trajectory. 
Here there is a very strong scale drift just before the loop closure, and the loop closing tries to distribute this error along the whole trajectory (as if the drift was uniform, but it was not!!).

That strong scale drift is produced by an almost pure rotation of the camera.