raulmur/ORB_SLAM #68: library issue

warsang, 2015-10-07:
Hello I am currently experiencing the following issue while trying to test ORB_SLAM with Ros Jade on ubuntu 15.04 
As I try to make the makefile generated by cmake for the ORB_SLAM I get the following errors:

/usr/bin/ld: cannot find -l:/usr/lib/x86_64-linux-gnu/libtinyxml.so
/usr/bin/ld: cannot find -l:/usr/lib/libPocoFoundation.so
collect2: error: ld returned 1 exit status
CMakeFiles/ORB_SLAM.dir/build.make:1500: recipe for target '../bin/ORB_SLAM' failed
make[2]: *** [../bin/ORB_SLAM] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/ORB_SLAM.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2

Can somebody please help? Thanks in advance

ladevan, 2015-10-24:
Hi,

Did you solve the problem? I have the same issue here. I am also using Ubuntu 15.04 and ROS Jade.

warsang, 2015-10-25:
Unfortunately no, but someone told me to try setting it up by installing the ROS indigo vm
http://nootrix.com/2014/09/ros-indigo-virtual-machine/
Didn't have time to try it out though :/

maym86, 2015-11-13:
Same issue here? Did anyone solve this:
[  0%] Built target rospack_genmsg_libexe
[  0%] Built target rosbuild_precompile
Linking CXX executable ../bin/ORB_SLAM
/usr/bin/ld: cannot find -l:/usr/lib/x86_64-linux-gnu/libtinyxml.so
/usr/bin/ld: cannot find -l:/usr/lib/libPocoFoundation.so
collect2: error: ld returned 1 exit status
CMakeFiles/ORB_SLAM.dir/build.make:1500: recipe for target '../bin/ORB_SLAM' failed
make[2]: *** [../bin/ORB_SLAM] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/ORB_SLAM.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2

wangzhaoguang, 2017-07-22:
same ! not gone through yet.