raulmur/ORB_SLAM #71: Stops somewhere in the middle

mojovski, 2015-10-14:
When I run ORB slam on this dataset
bag file: https://copy.com/ceBhVUN4vEgh9NtW
settings: https://copy.com/hYOBws1Y3lHq4k9t

the systems stops somewhere in the middle and does not continue.
I guess the reason is a fast rotation while walking.

Is there a way to debug this (I am new to ROS, sorry)?

mojovski, 2015-10-14:
by the way, the console output is:
```
Camera Parameters: 
- fx: 387.275
- fy: 387.275
- cx: 347.957
- cy: 273.672
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 10
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Fast Threshold: 20
- Score: FAST

Motion Model: Enabled

[ INFO] [1444825669.810379986]: New Map created with 78 points
[ INFO] [1444825669.813781254]: Wrong initialization, reseting...
[ INFO] [1444825670.208485079]: New Map created with 95 points
[ INFO] [1444825670.212561223]: Wrong initialization, reseting...
[ INFO] [1444825670.619745405]: New Map created with 94 points
[ INFO] [1444825670.623258352]: Wrong initialization, reseting...
[ INFO] [1444825671.021509974]: New Map created with 118 points

```

and then nothing more. From the code, it might be a infinite mutex wait...

gitunit, 2015-10-15:
@mojovski yes, i think it's due to your fast rotation to the right. however if you run your bag in a loop it will continue at the second run because it has more "knowledge" about the environment and therefore capable to track better. you could try from scratch with a slower motion, i bet it will not stop there.

raulmur, 2015-10-15:
@mojovski there is not infinite mutex wait. The problem is that the camera rotates so fast, the tracking get lost. See that at the bottom of ORB_SLAM/Frame window it says "TRACK LOST. TRYING TO RELOCALIZE". 
@gitunit is right. Move the camera a bit slower and the tracking won't fail.

closed 2015-10-15 by raulmur

mojovski, 2015-10-15:
@gitunit  and @raulmur , thank you very much!

@raulmur , I dont have the message "trying to relocalize". :/

Since I am new to ROS, two questions arise from your answers:

1. Does ORB SLAM process every frame coming from ROS-bag? If yes, do you mean I have to move the camera slower physically or can I do something without to change the dataset?

2. Is there a way to debug ORB Slam with a debugger (gdb) being run in ROS?

Thanks again!

raulmur, 2015-10-15:
Do you see the window with the current processed frame and tracked points?? (There is where you should see the message "Track Lost").

1. ORB-SLAM has 3 threads: Tracking, Mapping and Loop Closing. The Tracking processes frames as they come from the rosbag, but it may skip some if it is not able to process them at frame-rate. On the other hand the Mapping creates new points and perform Bundle Adjustment with every new keyframe.
If the movement is too fast, even if Tracking is able to process every frame, the Mapping may not be able to create new points fast enough so that Tracking can use them.

If you play the rosbag slower "rosbag play IPS_SHORT_RUN.bag -r 0.7" the system process the whole sequence (at least in my computer). Without a deeper analysis I do not know if the problem is frame skip by the Tracking or the Mapping is not able to expand the map fast enough. This solution wouldn't work in an online operation, that's why you should move the camera slower (physically).

2. You can use qtcreator to debug.

mojovski, 2015-10-15:
ok, thank you very much!
I found now this explanation about how to debug in ROS with qtcreator.

I guess its not possible to force processing of every frame in ROS even if its not real time?

mojovski, 2015-10-19:
Hi @raulmur,

just wanted to say that the trick with the playback speed reduction worked.
It also worked to increase the roshandle [buffer queue size](http://docs.ros.org/jade/api/roscpp/html/classros_1_1NodeHandle.html#aad44a59900a86f6d24f175b28e20c8a3).

I also faced an issue at another place of the dataset (not inside the published link) where the tracking is lost because of blur during a rotation. 
It helped to pre-process (sharpen) the images around the position where the track is lost. However, it is important not to sharpen all images as the slam becomes much worse and the tracking breakes down at another position.

Sofar, thanks again and I will leave this thread now in peace. :)