raulmur/ORB_SLAM #72: Can't run ORB-SLAM

SergioGarG, 2015-10-16:
Hello,

I'm having problems trying to run ORB-SLAM. I can't even run succesfully the launch file. I've installed it following the steps from the git (but i had to add the include #include <opencv2/opencv.hpp> to the ORBextractor.cc file because i couldn't do make in the root). 

When I run roslaunch ORB_SLAM ExampleGroovyOrNewer.launch the terminal tells me the following: 

sergio@sergio-MacBookPro:~/tfm/src/ORB_SLAM$ roslaunch ORB_SLAM ExampleGroovyOrNewer.launch 
... logging to /home/sergio/.ros/log/11616338-73e7-11e5-8215-0c4de99d84e9/roslaunch-sergio-MacBookPro-14005.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://sergio-MacBookPro:37961/

SUMMARY
========

PARAMETERS
 * /image_view/autosize: True
 * /rosdistro: indigo
 * /rosversion: 1.11.13

NODES
  /
    ORB_SLAM (ORB_SLAM/ORB_SLAM)
    image_view (image_view/image_view)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[image_view-1]: started with pid [14023]
process[rviz-2]: started with pid [14057]
process[ORB_SLAM-3]: started with pid [14080]

ORB-SLAM Copyright (C) 2014 Raul Mur-Artal
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.
[ORB_SLAM-3] process has died [pid 14080, exit code -11, cmd /home/sergio/tfm/src/ORB_SLAM/bin/ORB_SLAM Data/ORBvoc.txt Data/Settings.yaml __name:=ORB_SLAM __log:=/home/sergio/.ros/log/11616338-73e7-11e5-8215-0c4de99d84e9/ORB_SLAM-3.log].
log file: /home/sergio/.ros/log/11616338-73e7-11e5-8215-0c4de99d84e9/ORB_SLAM-3*.log

And if I play the bag file it starts but nothing happens.

Yes, the ORBvoc.txt.tar.gz file is uncompressed. I've even trying installing ORB SLAM three times.

If I try to run:

rosrun ORB_SLAM ORB_SLAM PATH_TO_VOCABULARY PATH_TO_SETTINGS_FILE

In my case:

rosrun ORB_SLAM ORB_SLAM Data/ORBvoc.txt Data/Settings.yaml

The terminal tells me the following: 

sergio@sergio-MacBookPro:~/tfm/src/ORB_SLAM$ rosrun ORB_SLAM ORB_SLAM Data/ORBvoc.txt Data/Settings.yaml

ORB-SLAM Copyright (C) 2014 Raul Mur-Artal
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.
Violación de segmento (`core' generado)
 (Not sure how to translate it, sorry)

Any help will be welcome, thanks!
Sergio

raulmur, 2015-10-16:
Without more information I cannot tell what is happening. Could you please use a debugger?

SergioGarG, 2015-10-19:
Hello again Raúl!

Due to I'm pretty new with ROS (and Ubuntu, to be fair) I had some troubles following your advice. I've tried with Eclipse, NetBeans and GDB trying to found the issue with the debugger. What is exactly what you want?

I can read from the log that the error that occurs in the launc is the next one:

[roslaunch][ERROR] 2015-10-19 13:03:17,326: [ORB_SLAM-4] process has died [pid 14208, exit code -11, cmd /home/sergio/tfm/src/ORB_SLAM/bin/ORB_SLAM /Data/ORBvoc.txt /Data/Settings.yaml __name:=ORB_SLAM __log:=/home/sergio/.ros/log/ff0b7a68-7650-11e5-ba3f-c8e0eb152e4d/ORB_SLAM-4.log].
log file: /home/sergio/.ros/log/ff0b7a68-7650-11e5-ba3f-c8e0eb152e4d/ORB_SLAM-4*.log

I've also tried to install it following the steps from here: 

https://github.com/srv/ORB_SLAM

But the terminal tells me back the following: 

sergio@sergio-MacBookPro:~/tfm/src/ORB_SLAM$ roslaunch ORB_SLAM ExampleGroovyOrNewer.launch 
... logging to /home/sergio/.ros/log/2f5d62b4-767c-11e5-b085-c8e0eb152e4d/roslaunch-sergio-MacBookPro-8546.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://sergio-MacBookPro:49619/

SUMMARY
========

PARAMETERS
 * /image_view/autosize: True
 * /rosdistro: indigo
 * /rosversion: 1.11.13

NODES
  /
    ORB_SLAM (ORB_SLAM/ORB_SLAM)
    image_view (image_view/image_view)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [8558]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 2f5d62b4-767c-11e5-b085-c8e0eb152e4d
process[rosout-1]: started with pid [8571]
started core service [/rosout]
process[image_view-2]: started with pid [8588]
process[rviz-3]: started with pid [8622]
ERROR: cannot launch node of type [ORB_SLAM/ORB_SLAM]: can't locate node [ORB_SLAM] in package [ORB_SLAM]

Thank you,
Sergio

SergioGarG, 2015-10-20:
I installed it in another PC (with Hydro) and it works, but I can't make it work in the other one with Indigo.

mhzaki, 2016-01-29:
Hi @SergioGarG  
I have the same problem as you but unfortunately it dosn't work with Hydro "ubuntu 12.04" can you help me please ?

SergioGarG, 2016-02-01:
Hi @mhzaki 

My problem was related with libraries (because I installed previously opencv2.4.8). So, I solved the problem uninstalling these libraries.

I don't know if your problem is related with it, I hope i helped you.

Sergio

selear129, 2016-04-22:
@SergioGarG  How do you solve the problem about the libraries?(I use ros-indigo  opencv 3.0 )