raulmur/ORB_SLAM #73: Question: How to store all frames?

mojovski, 2015-10-21:
I would like to extend ORB SLAM by an export method, which writes all keyframes and all normal frames to a file.
However, as far as I understand the code, only key frames are stored.

Right now I see here two possibilities:

**1st. Create from every frame a KeyFrame**
```
KeyFrame* pKF = new KeyFrame(mCurrentFrame,mpMap,mpKeyFrameDB);
```
and store is as a Child in the 
```
Tracking::mpLastKeyFrame->addChild(pKF);
```

**2nd: Extend the Keyframe by a list of Frames**
 and just append the current Frame to it

```
//Tracking class
std::vector<Frame> mIncludedFrames;
//and then
mIncludedFrames.push_back(mCurrentFrame); //or similar
```

As far as I see it now, the second approach would not allow to store also the pose of the frame, which is however required in the output export later.

Could someone advise me here?
The reason I am asking is that I would like to push the changes later into the main ORB_SLAM repository and dont want to mess up anything. :)

Thank alot!

jingpang, 2015-12-12:
@mojovski Hi, have you succeeded in saving keyframes to a file? Could you please share the method or some experience about saving the map?
I'd like to do the same thing as you now and I'm not sure about the method.

Thank you.

mojovski, 2015-12-13:
@jingpang I actually managed it to export only keyframes into an nvm file. However, there are still issues when I project a 3d point onto to neighbouring images the image coordinates are too different even is the camera not moving. So i expect still somewhere an issue in my code. I will push it these days to github and share it here, ok?

jingpang, 2015-12-16:
@mojovski Thank you for your reply! 
I found your work in another issue about saving data in nvm files, where I learned some method for my work.

I want to save the map (i.e. all keyframes and mappoints) in files to reload and reuse it next time, and it's a little different. But I'll really appreciate it if you can share your code.

VisualBarn, 2016-04-26:
I was also interested in the method used here to save the map  (i.e. all keyframes and mappoints) in files to reload and reuse it next time. Any help would be great!
Thank you!

stanchiang, 2016-09-12:
^bump and updates here? i'm interested in achieving the same functionality.

mojovski, 2016-09-12:
I managed it to export the keyframes to an nvm file some time ago. @stanchiang , would this help you? Moreover, I decided that I dont need to export all Frames.

stanchiang, 2016-09-12:
yes @mojovski I think it would help if i saw how you did your version as a starting point. My end goal is to use it with http://github.com/egoist-sx/ORB_SLAM_iOS. It is an iOS port of ORB_SLAM. Right now maps are not stored in any way. Ideally I want to load a previously created map and recognize where i am in the map with re-localization. This way I can build more robust maps without the need to do it all in one app session.

mojovski, 2016-09-12:
Hi @stanchiang , thanks for the link. Please have a look on this file: https://github.com/mojovski/ORB_SLAM/blob/export_nvm_2/src/main.cc (note the branch name: "export_nvm_2"). I had also somewhere a different version of the exporter. I will try to find it in a few days, when I have access again to my old computer.

as for the re-localization, this sounds very interesting. I am trying to find enough spare time to do experiments on this issue too. If you like, we can team up here.

stanchiang, 2016-09-13:
Thanks for the link. That will definitely save me time. Teaming up for this sounds like a great idea. Perhaps others in this tread would want to contribute as well?

mojovski, 2016-09-14:
@stanchiang just checked by working version of the code: its already pushed so the link I sent points to the newest version.

stanchiang, 2016-10-05:
@mojovski check out this thread. https://github.com/raulmur/ORB_SLAM2/issues/19

using the code from https://github.com/MathewDenny/ORB_SLAM2 save/load was very quick. i guess the next big advancement would be to implement semi-dense mode