raulmur/ORB_SLAM #75: Change Camera Source - ORB_Slam

Croninburg, 2015-11-04:
Hi Guys 

I have been looking into this and I seem to be stuck, all I am trying to do run this in real time, but I cant seem to change the camera source?? I am I missing something obvious ?? 

Any help would be greatly appreciated :) 

Chris

cipri-tom, 2016-02-21:
the node is watching on the `camera/image_raw` topic by default. if you have a node publishing images on some other topic, then you have to remap it to `camera/image_raw`.
For example, the `ardrone_autonomy` is publishing images on topic `/ardrone/front/image_raw`. To pass it to orb-slam I modified the lauch file to that in the `node` with ORB_SLAM I remap one topic onto the other:
```
    <node pkg="ORB_SLAM" .... output=screen>
        <remap from="/camera/image_raw" to="ardrone/front/image_raw" />
    </node>
```

KathiravanNatarajan, 2017-02-05:
@cipri-tom @Croninburg Hi, I am working on a school project. Do you guys have the source code of ORB_SLAM on ARDRONE 2.0 ? 

Any github link or suggestions are appreciable.

cipri-tom, 2017-02-06:
@KathiravanNatarajan you should spend time to understand how it works and plug it in yourself. What part are you stuck on ? Is there anything that's not working ?

KathiravanNatarajan, 2017-02-06:
I am getting this error when running the launch file 

`ERROR: cannot launch node of type [ORB_SLAM/ORB_SLAM]: can't locate node [ORB_SLAM] in package [ORB_SLAM]`

My package structure : 

catkin_ws 
...
ORB_SLAM    src   devel   etc
..(cloned files)

cipri-tom, 2017-02-09:
> Does bag file have anything to do with 3D model ?

Oh man, I really can't remember. And it was only 1 year ago :( . 
IIRC, the bag file was storing the visual words, i.e. an encoding of the environment which the algorithm can use to identify previously seen places.

With the existing code you cannot get the 3D model. All you're getting I think is a keyFrameTrajectory.txt which is not that useful.
I extended the project to export the map points at the closing of the program, in `PLY` format, which can be read by 3D processing software such as meshlab. However, keep in mind that you don't know the scale of the map, so you only get a rough idea of the relative locations. And there are outliers. Here are [2 images](http://imgur.com/a/v4eMs) of the 3D model I got out. Check my [project video](https://www.youtube.com/watch?v=NP1hGg2V7uM) and if you think it's useful, write me an email, so we don't spam this thread.