raulmur/ORB_SLAM #77: debug the code

Fahimkh, 2015-11-08:
Hi,

I want to run this code in debug mode.. I am trying so hard but still I am not able to do that.. I want to start with main file and look the code deeply.. Please help me to find the way...

raulmur, 2015-11-09:
I recommend you qtcreator. It can create a project just from the CMakeLists.txt

weblucas, 2015-11-09:
Indeed, qtcreator is the best for this. http://wiki.ros.org/IDEs

Fahimkh, 2015-11-10:
Thanks @raulmur  and @weblucas 
I followed the link but I am facing problem. I explain my problem

1-first I created a folder with name "ORB_WS" and in that folder I created another folder "sandbox".
2-then I downloaded the repository in "sandbox" folder.

my environment variables are like this
echo $ROS_PACKAGE_PATH
/home/faheem/ORB_WS/sandbox:/opt/ros/indigo/share:/opt/ros/indigo/stacks

echo $ROS_ROOT
/opt/ros/indigo/share/ros

3- I opened Qtcreator from terminal 
cd /home/faheem/Qt/Tools/QtCreator/bin
./qtcreator

4-After starting qtcreator I did following steps
    File --> Open File or Project...

5-I selected CMakeList.txt from ORB_SLAM folder

6-Then created a "build" folder in ORB_SLAM folder. I did this as I am following this link
http://wiki.ros.org/IDEs

7- I have attached all the snapshoots. After finishing I can see my source files.

Problem is that I can't see Thirdparty. When I put break point at the start of the main and run debug I  got error for Thirdparty. I dont know how to include this.

[image]

[image]

[image]

[image]

amiltonwong, 2015-11-10:
I had also followed steps from @Fahimkh , I also came across compile error (but a bit different) 
https://goo.gl/MFG4hM
It'll be greatly appreciated if one gives some suggestions~

Fahimkh, 2015-11-10:
Now I changed my method. But still facing one problem. I explain my working here

1-(same before)first I created a folder with name "ORB_WS" and in that folder I created another folder "sandbox".
2-(same before)then I downloaded the repository in "sandbox" folder.

my environment variables are like this
echo $ROS_PACKAGE_PATH
/home/faheem/ORB_WS/sandbox:/opt/ros/indigo/share:/opt/ros/indigo/stacks

echo $ROS_ROOT
/opt/ros/indigo/share/ros

3- (same before)I opened Qtcreator from terminal
cd /home/faheem/Qt/Tools/QtCreator/bin
./qtcreator

4-(same before)starting qtcreator

5a- File --> Open File or Project...
5b- First I opened g2o in Third party and added its CMakelist.txt
[image]

5c- Then I made a build folder in it and then I pressed "Run CMake" but with argument and I put this argument there ".. -DCMAKE_BUILD_TYPE=Debug"
[image]

6- I did this for DBoW2 and ORB_SLAM main folder. But I changed argument for ORB_SLAM main folder. In that case I put this argument ".. -DROS_BUILD_TYPE=Debug"

[image]

7- After that I double click on main.cc to open it in window. I put a break point and run the debug mode. I got this error

"make[3]: *** No rule to make target `../Thirdparty/DBoW2/lib/libDBoW2.so', needed by `../bin/ORB_SLAM'.  Stop.
make[2]: *** [CMakeFiles/ORB_SLAM.dir/all] Error 2
make[1]: *** [CMakeFiles/ORB_SLAM.dir/rule] Error 2
make: *** [ORB_SLAM] Error 2
23:27:46: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project ORB_SLAM (kit: Desktop Qt 5.3 GCC 64bit)
When executing step "Make""
[image]

@raulmur  @weblucas  please look on...

@amiltonwong  you can try this.. and if you find some solution please let me know...

raulmur, 2015-11-10:
Build first DBoW2 and g2o then build ORB-SLAM

Fahimkh, 2015-11-10:
@raulmur  I did it but still getting same error.. I think there is some problem in arguments that I am providing. Can you please check those arguments in above comment.

raulmur, 2015-11-10:
If I am not wrong the error in your last post is that DBoW2 is not found when building ORB-SLAM. You mentioned you ran CMake for DBoW2 and g2o but you did not build them separately before building ORB-SLAM.

Fahimkh, 2015-11-10:
Now it is working... :-) ... Basically I was putting this argment for "Run CMake" ..".. -DROS_BUILD_TYPE=Debug" and for g2o and DBoW2... When I changed this argument to ".. -DROS_BUILD_TYPE=Release" it started working...

How can I provide path to vocabulary and settings file in debug mode.
I am really thankful to you @raulmur

Fahimkh, 2015-11-10:
Sorry @raulmur  I am disturbing you.. I did this as well..
on the left side there is project tab.. pressed it and enter command line arguments...

closed 2015-11-10 by raulmur