raulmur/ORB_SLAM #78: Import this project into Eclipse

Fahimkh, 2015-11-08:
I am trying to import this project into eclipse. I am taking help from this page to make eclipse project.
http://wiki.ros.org/IDEs#Installing_Eclipse
Still I am not able to do that. Please help me I want to understand your project and do it more on it. It is really interesting.

mojovski, 2015-11-08:
Basically, most IDEs rely on a valid make file. However, when cmake is used, it can generate makefiles and project files for several IDEs. The most popular IDEs are Eclipse, code blocks, qt creator (used by the developers of ORB-SLAM). I prefer to use sublime text with the [GDB extension](https://thenigh7sh4de.wordpress.com/2014/03/28/setting-up-sublimegdb/).

There is also a new alternative: [CLion](https://www.jetbrains.com/clion/). I tried it a few days ago  and it works very fluently. However, its not free.

Fahimkh, 2015-11-08:
@mojovski 
Can you help me to build that project files for eclipse.. when I use the process as describe on this project page. I dont find any project file.t

I read about this command as well
cmake -G"Eclipse CDT4 - Unix Makefiles"

and applied it as well but still it is not working....

Fahimkh, 2015-11-08:
When I run this process at very start 

mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release
make 

I even dont find .project and .cproject file there... It even dont exist in build folder...

boonjiashen, 2016-04-02:
@Fahimkh I'm also trying to build ORB_SLAM on Eclipse to more easily familiarize myself with the codebase (I usually use Vim and a compiler). I haven't gotten the code to build, but I think I'm close. Here's my steps so far:

1. **Download the latest Eclipse CDT**, which is Eclipse 4.5.2 from [Eclipse](https://eclipse.org/cdt/). I'm using this version instead of Eclipse 3.8 (Juno) which is the default in the Ubuntu repo for Ubuntu 14.04, because I had trouble installing the Callgraph viewer plugin in Eclipse 3.8 while it was easy with 4.5.2 
2. **Create Eclipse project files with CMake**

```
#From ORB_SLAM root directory
mkdir build
cd build
cmake -G"Eclipse CDT4 - Unix Makefiles" -DCMAKE_BUILD_TYPE=Release ..
```

You should see a .project and .cproject in the build directory if you run `ls -a`

3. **Get Eclipse to find ROS header files**. If you don't do this step you'll get squiggly lines below your ros imports in Eclipse. This one is simple. From a ROS-sourced terminal, run the Eclipse executable. [[source](http://wiki.ros.org/IDEs#Reusing_your_shell.27s_environment)]

4. **Get Eclipse to find Eigen header files**. If you don't do this step you'll get squiggly lines below your Eigen imports in Eclipse. For this step you'll have to create symbolic links to the Eigen headers. Follow steps [here](http://answers.ros.org/question/61407/standard-system-include-paths-missing-in-eclipse-from-catkin_make/?answer=75748#post-id-75748).

5. **Open project in Eclipse**. With Eclipse open, just go New > General > Project and change the location from default location to your ORB_SLAM top level directory.

At this point most of the importing problems are ironed out, but I can't get Eclipse to build the project. When I click Project > Build > Build Project, I get 'nothing to build for ORB_SLAM' or no make target, depending on my build settings. I've tried messing around with different build settings, and changing the recognized C++ source file extension from cpp to cc, but none of them have worked.

Any help would be greatly appreciated.

boonjiashen, 2016-04-02:
OK I figured out how to build the project. The problem was that Eclipse doesn't know where the Makefile is.

1. **Tell Eclipse where the Makefile is**. Go to Project > Properties > Builder Setting, uncheck 'Use default build command" and set "Build command" to be `bash - c "cd build && make && cd ..", which essentially tells Eclipse that the Makefile is in the build subdirectory.

2. **Set environment variables in Eclipse**. I found that the minimum you'll have to set is ROS_PACKAGE_PATH, otherwise Eclipse wouldn't be able to find ORB_SLAM, curiously enough. For that, go to Project > Properties > C/C++ Build > Environment, and set ROS_PACKAGE_PATH to its value in your terminal.

3. **Run Project > Build All**. This command should execute the same behavior as running make in the terminal.