raulmur/ORB_SLAM #79: Problems initializing tracking

Loeing, 2015-11-09:
Hi Raul,

I have been taking video from an Android device, breaking that video down into images and using those images to create a rosbag using your BagFromImages package. I have no issues playing the rosbag but ORB_SLAM is having trouble initializing. It will occasionally successfully start tracking and then lose tracking and be unable to relocalize.
The Example bag you provided works perfectly. I've compared it to the videos I've taken and cannot seem to find major differences ( other than size of the image ). I am creating a loop around non-moving objects, the scenes I am looking have are rich in features, and there aren't any pure rotations.
I saw your message about using qtcreator to debug the code, so I will try to use that but any advise would be appreciated.

Thank you in advance,
Josh

raulmur, 2015-11-09:
Have you calibrated your camera?

Loeing, 2015-11-09:
Ah, I have not. Thank you for your help.

closed 2015-11-09 by raulmur