raulmur/ORB_SLAM #81: Question about camera calibration

elega, 2015-11-17:
Hello,
I am trying to change the video source from the example package to  my own webcam.I have used the usb-cam package of ros and it works.But the matches of orb displayed in the screen of imageview is very little and the track frequently lost .I guess there are some difference in parameter of webcam between ours and yours.I have used some methods to calibration my webcam,but the effect is not good.
I'd like to know how do you calibrate your camera,by ros package or opencv or other methods, and if the calibration results can directly be used in the file "Settings.yaml" .

Any help will be welcome,thank you.

Loeing, 2015-11-17:
Hi,
I was having a similar problem but managed to fix it. Here's what worked for me: I used the opencv  calibration using the asymmetrical circle pattern. I replaced the first 8 fields of the Settings.yaml in Data with the results of the calibration.  

Hope that helps a bit.

elega, 2015-11-17:
I used the opencv calibration using the chess board pattern.Maybe the asymmetrical circle pattern can make some difference.

Thank you very much.