raulmur/ORB_SLAM #88: Why the result of KITTI is worse than presented in the paper

ShichunYi, 2015-12-08:
Hi,
     I ran orbslam in catkin workspace and tested some sequences of KITTI. However, most of the results were worse than those on the paper. By the way, I have modified "Settings.yaml" on camera parameters.
Does anybody came across this problem ever?
     A reference camera path result compared with ground truth is:
    http://pan.baidu.com/s/1hrkTDAG (download link, sorry I failed to upload image here)

amiltonwong, 2015-12-08:
What configurations does your machine have?
The baseline is i7 cpu

ShichunYi, 2015-12-08:
My laptop runs on i5 cpu. But I have added some code to make the program working with KITTI image sequences, which means each frame will be processed. Does the computer configuration matter in this condition?

amiltonwong, 2015-12-09:
See:
https://github.com/raulmur/ORB_SLAM/issues/2#issuecomment-83440067

ShichunYi, 2015-12-09:
@amiltonwong , thanks very much for you reply. I read the #2 comment but I think my problem is different. They came across with "LOST TRACKING" failure because of weak cpu ( i3 ) and suggested rosbag test. However, I tested the image sequence frame by frame without transferring the images to .rosbag file. No matter how long it takes to tracking a new frame, the consecutive frame won't be drop out.
 
I didn't change the other two threads( LocalMapping and LoogClosing ), does it matters?

ps: I used the following function for image sequence test which will be called in Tracking::Run();
void Tracking::simuWithSeqImg(std::string imgdir, std::string timefile){
	double curtime;
	char name[200];
	int count = 0;
	std::ifstream timein(timefile.c_str());

	ros::Rate r(5);
	while( ros::ok() ){
		double time;
		std::sprintf(name, "%06d.png", count++);
		std::string imgname = imgdir + "/" + std::string(name);
		cv::Mat im = cv::imread(imgname, cv::IMREAD_GRAYSCALE);
		if (im.empty() )
			break;
		timein>>time;
                // process same as function "GrabImage()"
                r.sleep();
          }
 }