raulmur/ORB_SLAM #89: Error reading YAML file from example. Maybe OpenCV library conflict?

nlw0, 2015-12-08:
Hi. So I am using Ubuntu 15.04, with the official ROS packages, and I am trying to build this package. Eventually I decided I should move to a local version of OpenCV 3.0.0 that I built, and after building orb_slam I have this problem when I try to launch the example:

$ roslaunch ../ExampleGroovyOrNewer.launch 
... logging to /home/nlw/.ros/log/5169f6c2-9db6-11e5-87fc-d067e5f7cecc/roslaunch-kvelertak-22203.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://kvelertak:41533/

SUMMARY
========

PARAMETERS
 * /image_view/autosize: True
 * /rosdistro: jade
 * /rosversion: 1.11.16

NODES
  /
    ORB_SLAM (ORB_SLAM/ORB_SLAM)
    image_view (image_view/image_view)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[image_view-1]: started with pid [22280]
process[rviz-2]: started with pid [22287]
process[ORB_SLAM-3]: started with pid [22292]

ORB-SLAM Copyright (C) 2014 Raul Mur-Artal
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

[ERROR] [1449586588.150097151]: Wrong path to settings. Path must be absolut or relative to ORB_SLAM package directory.
[ORB_SLAM-3] process has died [pid 22292, exit code -11, cmd /home/nlw/src/ORB_SLAM/bin/ORB_SLAM Data/ORBvoc.txt Data/Settings.yaml __name:=ORB_SLAM __log:=/home/nlw/.ros/log/5169f6c2-9db6-11e5-87fc-d067e5f7cecc/ORB_SLAM-3.log].
log file: /home/nlw/.ros/log/5169f6c2-9db6-11e5-87fc-d067e5f7cecc/ORB_SLAM-3*.log
[rviz-2] process has died [pid 22287, exit code -11, cmd /opt/ros/jade/lib/rviz/rviz -d /home/nlw/src/ORB_SLAM/Data/rviz.rviz __name:=rviz __log:=/home/nlw/.ros/log/5169f6c2-9db6-11e5-87fc-d067e5f7cecc/rviz-2.log].
log file: /home/nlw/.ros/log/5169f6c2-9db6-11e5-87fc-d067e5f7cecc/rviz-2*.log

I believe this is related to my usage of this separate OpenCV library... So I was wondering if anyone can help me here. Should I maybe just try to build my own OpenCV, then ROS than ORB_SLAM? With what versions? Can it live along Ubuntu packages, or should I just start working in a virtual machine, for instance?

The reason I moved to this new OpenCV build was because of a "cv::" error when compiling ORB_SLAM, probably because it was trying to use an old OpenCV version, from when there was no namespace...

nlw0, 2015-12-09:
Just leaving this here in case it might help anyone. I ended up rebuilding my own OpenCV (3.0.0) and ROS (Indigo) from source and it worked.

quattrinili, 2016-01-19:
I have Ubuntu 14.04 with ROS Indigo and I installed also OpenCV 3.0 through the package available in the ROS repository ros-indigo-opencv3. By default, cmake links ORB_SLAM with OpenCV 3.0, however, the same error mentioned in the previous comment happens. It seems to be related to some changes in the FileStorage class in the new version of OpenCV that does not correctly read the yaml file.

A solution that worked for me is to use the OpenCV 2.4, which is the tested version for ORB_SLAM as reported in the README, and so explicitly write the 2.4 version for OpenCV in the CMakeLists.txt.