raulmur/ORB_SLAM #95: how to make KeyFrameTrajectory.txt graphical

a412841321, 2016-01-05:
Hi Raulmur
    I'm new to SLAM.I want to know how to make KeyFrameTrajectory.txt graphical.
   thank you!

cipri-tom, 2016-02-21:
hi @a412841321 ,
have you had any success ? Thanks!

a412841321, 2016-02-22:
Now I show KeyFrameTrajectory.txt with matlab. matlab can dispaly the Trajectory in three dimensional coordinate system. But the display isn't real-time.

cipri-tom, 2016-02-22:
Thanks @a412841321 ! Would you mind detailing that a bit ? I understood from [TUM](http://vision.in.tum.de/data/datasets/rgbd-dataset/file_formats) what the format is and I have an idea of how to plot the `t` component in matlab (camera position); basically just 3D plotting. Did you also display the orientation somehow ?

Many thanks!

VisualBarn, 2016-04-26:
How do you export KeyFrameTrajectory.txt from ORB-SLAM to be able to use it in Matlab?
Thank you!

chaiein, 2017-01-13:
Even I want to do it. How to make it graphical in matlab?