raulmur/ORB_SLAM #98: Unable to see any output from the rosbag file!

ank700, 2016-01-13:
I run the bag file, downloaded from this link (http://webdiis.unizar.es/%7Eraulmur/orbslam/downloads/Example.bag.tar.gz). Then on rviz I can only see a video of a camera being moved randomly in a room. Is this what is expected from this bag file or am I missing something. Any help is much appreciated. Thanks

qinmengsu, 2016-07-04:
Try:
cd /ORB_SLAM directory
roslaunch ExampleGroovyOrNewer.launch
rosbag play --pause Example.bag

I met also this problem same with you when I run step by step (ORB_SLAM, image view, rviz, rosbag play), but no problem when just do roslaunch like above.

trinamntn08, 2017-06-22:
I met the same problem. Do you have idea to resolve this issue?