raulmur/ORB_SLAM #99: Speed up the process which read vocabulary file and build tree

dganbold, 2016-01-20:
I am trying to execute the ORB-SLAM on raspberry pi2 for mobile robot navigation,but faced with two challenges first one is the loading vocabulary file and building tree is takes time, second is update rate of tracking is low.
Vocabulary loading time problem is already solved by implementing class serialization, 
particularly serialization and deserialization of "ORBVocabulary" class. It dramatically decreased about 30times,  3 minute go down to 6 second on rapsberry pi2 ( Ubuntu + ROS Indigo ), 15 sec to aroud 0.5 sec on PC which utilized Intel CORE-M.

Now, I am struggling with where increase the update rate to be the next challenge.

Loeing, 2016-01-20:
Hi dganbold,
Sorry I can't help with the update rate (I'm trying to solve the same problem) could you link me to your code for serializing and deserializing the ORBVocabulary? 
Thanks,
Josh

Beewe, 2016-02-01:
I added a speeded up function `loadFromTextFile2` for reading and parsing an updated ORBVocabulary to my fork of ORB_SLAM2:
https://github.com/Beewe/ORB_SLAM2_Windows/commit/bcd957c48c9ebc3bd4839bb5e9e036783faabd4f