raulmur/ORB_SLAM #100: windows version?

antithing, 2016-01-21:
Hi there, this looks amazing. Any chance that it would run on windows? Thanks!

Beewe, 2016-01-23:
Yes, I've already have it running under windows. I'm planning to release this version on github, but first I have to merge the recent updates from this repo to mine and make sure everything is working. Also I've made it working as a library, so you can call TrackFrame(cv::Mat frame) and receive the camera pose.

antithing, 2016-01-23:
Awesome! Thank you for all your work. :) I look forward to getting it
running.

Beewe, 2016-01-26:
Hey, as you've maybe already seen there is a new version called ORB_SLAM2, which includes several new features, so I decided to make this Windows compatible. Check out my fork, I've got it already running: https://github.com/Beewe/ORB_SLAM2_Windows
Let me know if you still need/want a ORB_SLAM v1.

antithing, 2016-01-26:
awesome. Thank you! I will check this out asap.

antithing, 2016-01-26:
Hi, I cloned this dir and built the solution, but am getting errors on the
visual studio build..

Severity Code Description Project File Line Suppression State
Error C2280 'Eigen::Block<Eigen::Block<Eigen::Map<Eigen::Matrix<double,-1,-1,0,-1,-1>,0,Eigen::Stride<0,0>>,-1,-1,false,true>,-1,-1,false,true>
&Eigen::Block<Eigen::Block<Eigen::Map<Eigen::Matrix<double,-1,-1,0,-1,-1>,0,Eigen::Stride<0,0>>,-1,-1,false,true>,-1,-1,false,true>::operator
=(const
Eigen::Block<Eigen::Block<Eigen::Map<Eigen::Matrix<double,-1,-1,0,-1,-1>,0,Eigen::Stride<0,0>>,-1,-1,false,true>,-1,-1,false,true>
&)': attempting to reference a deleted function ORB_SLAM2
c:\libs\eigen\include\eigen\src\Core\SolveTriangular.h 194

From this section of Eigen:

if (copy) other = otherCopy;

Do you have any idea what that would be?

Thanks.

Beewe, 2016-01-26:
Just having a guess: Maybe you use an outdated version of Eigen, see "Prerequisites" from the install instructions; you have to use Eigen at least in version 3.1.0. I just checked the linenumber 194 in version 3.0.7 and it has the same content as you described "if (copy) other = otherCopy;", whereas the linenumber 194 in version 3.2.4 (which I use) has a completly different content.

antithing, 2016-01-26:
Got it, I just noticed that instruction as I saw your message. Thanks!

antithing, 2016-01-27:
Heya, sorry for all these questions! I have the kitti example building, but
it is crashing on loading the Vocabulary file. I get to:

'Loading ORB Vocabulary. This could take a while...' Then the program
crashes with 'abort has been called'. In a related question, if I run this
live using a camera input, do I need the Vocabulary file? If so, how do I
create this?

Thanks!

Beewe, 2016-01-28:
You always need the Vocabulary file, it is used for place recognition and feature matching. There is already one in the Vocabulary directory which you you first have to extract because it's packed. Sorry I have never experienced such a crash yet. Did it crash immediately after "This could take a while..." or does the file load completly?

antithing, 2016-01-28:
'This could take a while...'  When I debug, it actually errors on this line:

  F::fromString(m_nodes[nid].descriptor, ssd.str());

in TemplatedVoccabulary.h.

Thanks!

antithing, 2016-01-29:
FIXED! I had some 64bit / 32bit dll conflicts. Rebuilding all sorted it.
Thanks again!