raulmur/ORB_SLAM #104: Error during make- for DBoW2

gunshi, 2016-04-22:
Output of cmake step:

> -- The C compiler identification is AppleClang 7.0.2.7000181
> -- The CXX compiler identification is AppleClang 7.0.2.7000181
> -- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
> -- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
> -- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Configuring done
> CMake Warning (dev):
>   Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
>   --help-policy CMP0042" for policy details.  Use the cmake_policy command to
>   set the policy and suppress this warning.
> 
>   MACOSX_RPATH is not specified for the following targets:
> 
>    DBoW2
> 
> This warning is for project developers.  Use -Wno-dev to suppress it.
> 
> -- Generating done
> -- Build files have been written to: /Users/gunshigupta/Desktop/4sfm/ORB_SLAM/ThirdParty/DBoW2/build

Error during make:
> 
> Scanning dependencies of target DBoW2
> [ 14%] Building CXX object CMakeFiles/DBoW2.dir/DBoW2/BowVector.cpp.o
> [ 28%] Building CXX object CMakeFiles/DBoW2.dir/DBoW2/FORB.cpp.o
> /Users/gunshigupta/Desktop/4sfm/ORB_SLAM/ThirdParty/DBoW2/DBoW2/FORB.cpp:16:10: fatal error: 
>       'stdint-gcc.h' file not found
> #include <stdint-gcc.h>
>          ^
> 1 error generated.
> make[2]: *** [CMakeFiles/DBoW2.dir/DBoW2/FORB.cpp.o] Error 1
> make[1]: *** [CMakeFiles/DBoW2.dir/all] Error 2
> make: *** [all] Error 2
> yguns-MacBook-Pro:build gunshigupta$ cc
> clang: error: no input files

coskunyerli, 2016-05-03:
you try to build with clang, so stdind-gcc.h not found in clang. Erase "-gcc" and put "stdint.h" instead of "stdint-gcc.h"

gunshi, 2016-05-03:
Resolved, thanks!

closed 2016-05-03 by gunshi