raulmur/ORB_SLAM #105: Error building g2o, modification for OSX

gunshi, 2016-04-22:
OSX: 10.11.12

Output of   cmake -stdlib=libc++ .. -DCMAKE_BUILD_TYPE=Release

> -- The C compiler identification is AppleClang 7.0.2.7000181
> -- The CXX compiler identification is AppleClang 7.0.2.7000181
> -- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
> -- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
> -- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- BUILD TYPE:Release
> -- Compiling on Unix
> -- Looking for sgemm_
> -- Looking for sgemm_ - found
> -- A library with BLAS API found.
> -- Looking for cheev_
> -- Looking for cheev_ - found
> -- A library with LAPACK API found.
> -- Try OpenMP C flag = [-fopenmp=libomp]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Failed
> -- Try OpenMP C flag = [ ]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Failed
> -- Try OpenMP C flag = [-fopenmp]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Failed
> -- Try OpenMP C flag = [/openmp]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Failed
> -- Try OpenMP C flag = [-Qopenmp]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Failed
> -- Try OpenMP C flag = [-openmp]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Failed
> -- Try OpenMP C flag = [-xopenmp]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Failed
> -- Try OpenMP C flag = [+Oopenmp]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Failed
> -- Try OpenMP C flag = [-qsmp]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Failed
> -- Try OpenMP C flag = [-mp]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Failed
> -- Try OpenMP CXX flag = [-fopenmp=libomp]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Failed
> -- Try OpenMP CXX flag = [ ]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Failed
> -- Try OpenMP CXX flag = [-fopenmp]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Failed
> -- Try OpenMP CXX flag = [/openmp]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Failed
> -- Try OpenMP CXX flag = [-Qopenmp]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Failed
> -- Try OpenMP CXX flag = [-openmp]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Failed
> -- Try OpenMP CXX flag = [-xopenmp]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Failed
> -- Try OpenMP CXX flag = [+Oopenmp]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Failed
> -- Try OpenMP CXX flag = [-qsmp]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Failed
> -- Try OpenMP CXX flag = [-mp]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Failed
> -- Could NOT find OpenMP (missing:  OpenMP_C_FLAGS OpenMP_CXX_FLAGS) 
> -- Found Eigen3: /usr/local/include/eigen3 (Required is at least version "3.1.0") 
> -- Configuring done
> CMake Warning (dev):
>   Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
>   --help-policy CMP0042" for policy details.  Use the cmake_policy command to
>   set the policy and suppress this warning.
> 
>   MACOSX_RPATH is not specified for the following targets:
> 
>    g2o
> 
> This warning is for project developers.  Use -Wno-dev to suppress it.
> 
> -- Generating done
> -- Build files have been written to: /Users/gunshigupta/Desktop/4sfm/ORB_SLAM/Thirdparty/g2O/build

Output of make:

> Scanning dependencies of target g2o
> [  3%] Building CXX object CMakeFiles/g2o.dir/g2o/types/types_sba.cpp.o
> In file included from /Users/gunshigupta/Desktop/4sfm/ORB_SLAM/Thirdparty/g2O/g2o/types/types_sba.cpp:27:
> In file included from /Users/gunshigupta/Desktop/4sfm/ORB_SLAM/Thirdparty/g2O/g2o/types/types_sba.h:30:
> In file included from /Users/gunshigupta/Desktop/4sfm/ORB_SLAM/Thirdparty/g2O/g2o/types/../core/base_vertex.h:30:
> In file included from /Users/gunshigupta/Desktop/4sfm/ORB_SLAM/Thirdparty/g2O/g2o/types/../core/optimizable_graph.h:38:
> /Users/gunshigupta/Desktop/4sfm/ORB_SLAM/Thirdparty/g2O/g2o/types/../core/hyper_graph.h:41:10: fatal error: 
>       'tr1/unordered_map' file not found
> #include <tr1/unordered_map>
>          ^
> 1 error generated.
> make[2]: *** [CMakeFiles/g2o.dir/g2o/types/types_sba.cpp.o] Error 1
> make[1]: *** [CMakeFiles/g2o.dir/all] Error 2
> make: *** [all] Error 2

I added 

> set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -stdlib=libc++")
tot he cmakelists.txt, but i still get the same error.

coskunyerli, 2016-05-03:
not found boost library 
you must add boost library your g2o cmake file

build with xcode. it is easier. you can add boost library path easily in xcode

gunshi, 2016-05-03:
Hi, but this error seems connected to the tr1 namespace and headers issue on OS X, doesn't it?

coskunyerli, 2016-05-03:
you can add to cmake boost library or you can build with xcode and add to search header path /usr/local/boost/tr1 and /usr/local/boost

closed 2016-05-04 by gunshi

cduguet, 2016-05-19:
How did you solve it, @gunshi ??? I'm having the same problem

gunshi, 2016-05-21:
Hi, I followed the instructions here, as per my conversation in the comments on this commit page
https://github.com/dacuda/g2o/commit/95dc775dd9e7234d46698cf39e6617ffea7955c0

gunshi, 2016-05-22:
Hi, just follow the instructions as per my comments on the following commit page:
https://github.com/dacuda/g2o/commit/95dc775dd9e7234d46698cf39e6617ffea7955c0
You could omit the changes related to OpenGL if you want gl support.